robotology/yarp #1989: Failure in finding manager YARP component

traversaro, 2019-04-01:
**Describe the bug**
A project that contains:
~~~
find_package(YARP 3.0 COMPONENTS manager REQUIRED)
~~~
fails to configure with the latest YARP master branch, with the following error:
~~~
CMake Error at /home/straversaro/src/robotology-superbuild/build-clean/install/lib/cmake/YARP/YARPConfig.cmake:183 (find_package):
  Found package configuration file:

    /home/straversaro/src/robotology-superbuild/build-clean/install/lib/cmake/YARP_manager/YARP_managerConfig.cmake

  but it set YARP_manager_FOUND to FALSE so package "YARP_manager" is
  considered to be NOT FOUND.  Reason given by package:

  The following imported targets are referenced, but are missing:
  YARP::YARP_sig YARP::YARP_math

Call Stack (most recent call first):
  CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/straversaro/src/robotology-superbuild/build-clean/robotology/icub-tests/CMakeFiles/CMakeOutput.log".
~~~

**To Reproduce**
For an example project, try to configure the current icub-tests master branch against YARP master branch.

**Expected behavior**
The configuration should work correctly.

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04 
 - yarp version: https://github.com/robotology/yarp/commit/69fd955227644d935a138490d1e45d0edbbed93e
-  compiler: gcc 7.3

**Additional context**
Related robotology-superbuild issue: https://github.com/robotology/robotology-superbuild/issues/192 .

traversaro, 2019-04-01:
A possible workaround is to manually search the required component dependency before the component actually required: 
~~~
find_package(YARP 3.0 COMPONENTS math manager REQUIRED)
~~~

traversaro, 2019-04-01:
It is possible that the PRIVATE_DEPENDENCIES argument logic in https://github.com/robotology/yarp/blob/master/src/libYARP_manager/CMakeLists.txt#L129 is broken?

traversaro, 2019-04-01:
I did not check, but I guess the issue could be due to the fact that I use YCM 0.10 .

drdanz, 2019-04-01:
If the project is icub-tests, it is fixed in the `robottestingframework-2` branch. The manager component is not used. The manager component is built and installed only if the yarpmanager or yarpmanager-console are built, are you sure they are compiled in the superbuild?

diegoferigo, 2019-04-02:
I encountered the same error, maybe [this (pretty long) log](https://api.travis-ci.com/v3/job/189233844/log.txt) might help. Just reach the end where it fails.

Here the related [CMake configuration](https://github.com/diegoferigo/development-iit/blob/a6a6e1753fbbf4f13a69de39e227b1a59c5ffbcc/Development/Dockerfile#L379-L395).

traversaro, 2019-04-02:
@diegoferigo sorry, I  forgot to forward merge https://github.com/robotology/icub-tests/pull/46 to devel.

traversaro, 2019-04-02:
Should have been fixed (in icub-tests) by https://github.com/robotology/icub-tests/commit/b05df9753ccd1963c83cb0066cb8064e8f5ad673 .

diegoferigo, 2019-04-03:
Yes, it seems fixed, great! Thank you

Nicogene, 2019-04-30:
> Should have been fixed (in icub-tests) by robotology/icub-tests@b05df97 .

Shall we close this issue?

drdanz, 2019-04-30:
I think we should still test and eventually fix this

traversaro, 2021-04-20:
I tested now with YARP 3.4.3 and the issue is still there, if I configure: 
~~~
cmake_minimum_required(VERSION 3.12)
project (hello)
find_package(YARP 3.0 COMPONENTS manager REQUIRED)
~~~

It fails with:
~~~
(yarp1989) traversaro@IITICUBLAP102:~/yarp1989_test/build$ cmake .
CMake Error at /home/traversaro/miniforge3/envs/yarp1989/lib/cmake/YARP/YARPConfig.cmake:162 (find_package):
  Found package configuration file:

    /home/traversaro/miniforge3/envs/yarp1989/lib/cmake/YARP_manager/YARP_managerConfig.cmake

  but it set YARP_manager_FOUND to FALSE so package "YARP_manager" is
  considered to be NOT FOUND.  Reason given by package:

  The following imported targets are referenced, but are missing:
  YARP::YARP_sig YARP::YARP_math

Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/traversaro/yarp1989_test/build/CMakeFiles/CMakeOutput.log".
~~~

the workaround is simply to list the transitive dependencies, i.e. this works fine:
~~~
cmake_minimum_required(VERSION 3.12)
project (hello)
find_package(YARP 3.0 COMPONENTS math sig manager REQUIRED)
~~~

@Nicogene not sure why you assigned this to me back in time (probably you assumed that it was fixed), but I think I can remove myself from the assignee list.

traversaro, 2022-06-24:
This issue has the same origin of https://github.com/robotology/yarp/issues/2861#issuecomment-1165438123 . In a nutshell, the problem is that `libYARP_manager` is hardcoded to be a static library.