robotology/yarp #2018: Deprecate Chicken support?

traversaro, 2019-05-11:
The recently release SWIG 4 ( https://sourceforge.net/p/swig/news/2019/04/swig-400-released/ ) dropped support for Chicken, that is explicitly supported in YARP build system. I guess this means that we should start deprecating the YARP-Chicken bindings.

drdanz, 2019-05-13:
+1 From me, probably the same should be done for allegrocl since it was removed as well.

Nicogene, 2019-05-13:
+1 also for for me, we should support the languages actually used, I'm pretty sure that the bindings of `chicken`, `allegro` and `tcl` are used by nobody.

drdanz, 2019-05-13:
I opened #2019, that just removes both languages, without actually deprecating them, I don't think it is worth to waste time on them...

traversaro, 2019-05-13:
If somebody actually complains we can always add them back easily for a proper deprecation cycle. : D

drdanz, 2019-05-13:
> +1 also for for me, we should support the languages actually used, I'm pretty sure that the bindings of chicken, allegro and tcl are used by nobody.

TCL bindings might be unused, but at least they are tested on travis :wink:

Nicogene, 2019-05-14:
Closed by #2019

PR #2019 (2019-05-13): Remove Chicken Scheme and Allegro Common Lisp bindings
These have not been used by anyone for a long time and are no longer supported in SWIG-4.0, see https://sourceforge.net/p/swig/news/2019/04/swig-400-released/

Fixes #2018
commit 41c4100f (2019-05-13): Remove Chicken Scheme and Allegro Common Lisp bindings

These have not been used by anyone for a long time and are no longer
supported in SWIG-4.0, see
https://sourceforge.net/p/swig/news/2019/04/swig-400-released/

Fixes #2018
closed 2019-05-14 by Nicogene