robotology/yarp #2020: IPositionDirect doesn't include what it uses

Nicogene, 2019-05-14:
**Describe the bug**
Any executable that include only `IPositionDirect.h` doesn't compile. 

**Expected behavior**
It should include `Vocab.h` and compile.

PeterBowman, 2019-05-14:
Might be already solved in devel thanks to https://github.com/robotology/yarp/pull/2006, see https://github.com/robotology/yarp/commit/864c3f851d36ad28b5d5dded442ef857477ffcb6#diff-fbcecfd8df1807946df81405a4ab316a.

Nicogene, 2019-05-14:
> Might be already solved in devel thanks to #2006, see e.g. 864c3f8#diff-fbcecfd8df1807946df81405a4ab316a.

:+1: 

I opened it for people like me that have a robot on `master` and maybe are experiencing this issue. Closing after the release.

closed 2019-06-27 by drdanz