robotology/yarp #2022: Unable to run yarprun from yarpmanager

GiulioRomualdi, 2019-05-15:
I was trying to run `yarprun`  using `yarpmanager`. When I click on `Run Selected`, the manager displayed the following message 
```
[MSG] ClusterWidget: yarprun successfully executed on icub-virtualizer
```
however, `yarprun` was not running.

It is important to notice that this happen only if the computer on which `yarprun` is launched is a Windows Machine. I've also tried to open an `ssh` instance from the computer in which `yarpmanager` is running to the windows machine and I tried to run `yarprun` from the terminal. In this specific case, everything goes well.

YARP was updated to the latest version (`devel` branch).

traversaro, 2019-05-15:
I do not think at the moment `yarpmanager` supports starting `yarprun` in Windows hosts, @Nicogene probably knows more about this. However, the "successfully executed" message should be different nad clearly specify that it did not work successfully.

Nicogene, 2019-05-15:
> I've also tried to open an ssh instance from the computer in which yarpmanager is running to the windows machine and I tried to run yarprun from the terminal. In this specific case, everything goes well.

If I understood correctly you ran from a linux machine to the Windows machine:
```
ssh user@host "yarprun --server /foo"
```
And it worked, right?

Have you checked if `yarprun` was running on windows machine despite the `ClusterWidget` shows it with a red icon?

Checking with `ps aux` or whatever windows has :grimacing: 

I honestly don't know how works ssh on windows, I know that is something that it has been introduced recently, but I don't know if has any issues.