robotology/yarp #2027: Twofold interpretation of pointer parameter in checkMotionDone() reflected in docs and implementations

PeterBowman, 2019-05-22:
https://github.com/robotology/yarp/blob/86cfea57195a756fd08df5bc73757ad87fbdffe8/src/libYARP_dev/include/yarp/dev/IPositionControl.h#L80-L85

The output parameter is described as:

> `flag` is a pointer to return value ("and" of all joints)

It is implemented as such in [ref1](https://github.com/robotology/yarp/blob/a211f01990c5de5845f37cc12aea2e09e53febd9/src/libYARP_dev/src/devices/RemoteControlBoard/RemoteControlBoard.cpp#L2018-L2020), [ref2](https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/src/libYARP_dev/src/devices/ControlBoardWrapper/RPCMessagesParser.cpp#L2412-L2418), [ref3](https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/src/libYARP_dev/src/devices/ControlBoardRemapper/ControlBoardRemapper.cpp#L1227-L1257) and [ref4](https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/bindings/yarp.i#L848-L853).

However, this leads to an asymmetry with regard to other motor-interface methods that accept a single pointer. In contrast to the previous behavior, sometimes it is regarded by the implementation as a pointer to an array: [ref1](https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/src/libYARP_dev/src/ImplementPositionControl.cpp#L161-L171), [ref2](https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/src/devices/DynamixelAX12Ftdi/DynamixelAX12FtdiDriver.cpp#L450-L457), [ref3](https://github.com/robotology/yarp/blob/b173764311dc0537a54c0bd6f58d808b5e415f4f/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L1769-L1800).

Unless I'm wrong, notice that ControlBoardWrapper's RPC parser ([ref](https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/src/libYARP_dev/src/devices/ControlBoardWrapper/RPCMessagesParser.cpp#L2412-L2418)) expects a single boolean value, whereas its implementation expects an array of booleans ([ref](https://github.com/robotology/yarp/blob/b173764311dc0537a54c0bd6f58d808b5e415f4f/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L1769-L1800)). Memory space outside the boundaries of the declared boolean variable is accessed through the `flag[juser] = done[jdevice];` instruction.

Related issues: https://github.com/robotology/yarp/issues/390, https://github.com/robotology/yarp/issues/387.

YARP version: current master branch.

PeterBowman, 2019-05-30:
TODO (targeting devel branch as of https://github.com/robotology/yarp/commit/fc929769ddf3a5097568ec679b4ac154ab5d562d):

- [fakeMotionControl.cpp](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/devices/fakeMotionControl/fakeMotionControl.cpp#L1968-L1983)
- [ControlBoardWrapper.h](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.h#L525-L531) (docs)
- [ControlBoardRemapper.cpp](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/libYARP_dev/src/devices/ControlBoardRemapper/ControlBoardRemapper.cpp#L1227-L1257)
- [RemoteControlBoard.cpp](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/libYARP_dev/src/devices/RemoteControlBoard/RemoteControlBoard.cpp#L2013-L2020)
- [RPCMessagesParser.cpp](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/libYARP_dev/src/devices/ControlBoardWrapper/RPCMessagesParser.cpp#L2412-L2418)
- IPositionControl.h: [1](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/libYARP_dev/include/yarp/dev/IPositionControl.h#L80-L85) + [2](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/libYARP_dev/include/yarp/dev/IPositionControl.h#L307-L312) (docs)

Also, list of *correct* implementations:

- [DynamixelAX12FtdiDriver.cpp](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/devices/DynamixelAX12Ftdi/DynamixelAX12FtdiDriver.cpp#L450-L457)
- [ControlBoardWrapper.cpp](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L1764-L1800)
- [ImplementPositionControl.cpp](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/libYARP_dev/src/ImplementPositionControl.cpp#L193-L206)

PeterBowman, 2019-05-30:
Also affects joint-group commands, e.g. [`RemoteControlBoard::checkMotionDone(int, int*, bool*)`](https://github.com/robotology/yarp/blob/fc929769ddf3a5097568ec679b4ac154ab5d562d/src/libYARP_dev/src/devices/RemoteControlBoard/RemoteControlBoard.cpp#L2003-L2011). Note that RPC protocol changes (bump protocol version).

barbalberto, 2019-05-31:
The correct interpretation is the one stated in the documentation: the method `checkMotionDone` is _**always returning a single boolean value**_  for all the APIs, like shown in the link you mentioned.

> However, this leads to an asymmetry with regard to other motor-interface methods 

I know this may be strange if compared to other methods and may lead a developer into error, as you say, but this behavior is documented, so IMHO is acceptable.

> Unless I'm wrong, notice that ControlBoardWrapper's RPC parser (ref) expects a single boolean value, whereas its implementation expects an array of booleans (ref).

Actually here you correctly spotted a regression bug introduced in 0ec738a9a33bf8394ac7e84209ae5f6d16e26ed6

Before that commit, all the three implementation of `checkMotionDone` were correctly returning a single value , see  https://github.com/robotology/yarp/blob/09fd2d76e7fd0db786277dcf3d9ced35342dfe6f/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L1731-L1754
Then a refactoring for optimization introduced the regression bug. Thanks for finding it out, I'll fix it.

About the implementation you linked:

<table>
  <tr>
    <td>fakeMotionControl.cpp </td>
    <td>all calls returns a single value: correct</td>
  </tr>
  <tr>
    <td>ControlBoardWrapper.h (docs) </td>
    <td> docs is correct </td>
  </tr>
  <tr>
    <td>ControlBoardRemapper.cpp </td>
    <td> checkMotionDone(bool *flag) is wrong </td>
  </tr>
  <tr>
    <td>RemoteControlBoard.cpp </td>
    <td>all calls returns a single value: correct</td>
  </tr>
  <tr>
    <td>RPCMessagesParser.cpp </td>
    <td>all calls returns a single value: correct</td>
  </tr>
  <tr>
    <td>IPositionControl.h: 1 + 2 (docs) </td>
    <td> doc is correct</td>
  </tr>
  <tr>
    <td>DynamixelAX12FtdiDriver.cpp </td>
    <td> multiple values: WRONG</td>
  </tr>
  <tr>
    <td>ImplementPositionControl.cpp </td>
    <td> same regression bug, WRONG </td>
  </tr>
</table>

So, to make the story short, there is no twofold interpretation, the only correct interpretation is the one explicitly stated in the documentation. There are some wrong implementation not complying with the documentation.

I'll fix those three implementation to comply with API.

valegagge, 2019-05-31:
I also checked icub motion control devices, both can and ethernet; the last one has the bug you noticed.
I'll fix it asap.

We need to fix the bug on yarp and on iCub device concurrently. If I'm not wrong the best way should be before fix iCub and  than yarp.

drdanz, 2019-06-03:
Fixed by @barbalberto and @valegagge in #2035 and robotology/icub-main#580

closed 2019-06-03 by drdanz