robotology/yarp #2036: yarpscope does not start

windelbouwman, 2019-06-03:
**Describe the bug**
When I startup yarpscope.exe, a blank window appears, and dies shortly after. I do not see any error messages or a message on the console.

Is there any way to provide more feedback about what went wrong?

**To Reproduce**
Steps to reproduce the behavior:
- Start yarpscope.exe from command line.

**Expected behavior**
Either an error message, or a functional yarpscope GUI.

**Configuration (please complete the following information):**
 - OS: Windows 10
 - yarp version: 3.1.0 64 bits

**Additional context**
Before installing YARP 3.1.0 64 bits, I installed the 32 bits version as well. The YARP environment variables both point to the 64 bit program files directory, so this should be good.

Nicogene, 2019-06-04:
Hi @windelbouwman ,

I tried `yarpscope` with yarp 3.1.101(actual `devel`) and it works fine on windows.

Did you installed `yarp` from binaries?
Are you sure to have run `yarpserver` before running `yarpscope`?

Nicogene, 2019-06-21:
I think this is not a YARP issue, but rather a broken installation of Qt, closing for now, feel free to re-open it if you find out some other problem.

closed 2019-06-21 by Nicogene

windelbouwman, 2019-06-21:
Hi @Nicogene, thank you for your reply. I uninstalled YARP. I think it was indeed an installation issue with some dll-hell thingy. I installed both 32 and 64 bits version, so that probably got bitten somewhere inside.

Nicogene, 2019-06-21:
A common error under windows is to have yarp compiled 32 bit that links Qt compiled for 64bit or viceversa.

For the future I advise to use [dependency walker](http://www.dependencywalker.com/) to see if you are linking the correct libraries.