robotology/yarp #2037: Unknown CMake command "swig_add_library" error at YARP 3.1.101

PeterBowman, 2019-06-03:
Originating from https://github.com/robotology/yarp/commit/a6d25e97dc5249003859a78dc909cadccd89d344 (import YCM 0.10.3), and as described in https://github.com/robotology/yarp/pull/2031#issuecomment-498321674, YARP is unable to find `swig_add_library()` via YCM modules whenever building language bindings. Happens on CMake configure step, mark `YARP_COMPILE_BINDINGS` and e.g. `CREATE_PYTHON`. Make sure that YCM was not previously installed or it has not been upgraded to release 0.10.3 (so that YARP picks its `extern/ycm` instead of the system-wide available YCM).

Proposed solution: expand the following check so that all `/extern/ycm/ycm-0.10.3/cmake-3.xx/Modules` directories are explored by CMake.

https://github.com/robotology/yarp/blob/e2c339713e554d487f3c99616de817ddd0022269/extern/ycm/CMakeLists.txt#L15-L17

Note that current YARP master (includes YCM 0.9) ignores `cmake-3.7` and `cmake-3.11` directories. Not sure if intended, these modules are consumed differently, or they are not needed at all.

Somewhat related: https://github.com/robotology/yarp/issues/1552.

**Configuration:**
 - OS: Ubuntu Xenial (16.04)
 - CMake: 3.5.1
 - yarp version: 3.1.101 (https://github.com/robotology/yarp/commit/ebde91ff9bf31b013bd6e2dd39dcbbef4a9b7a59)
   - note: latest devel branch fails to compile due to https://github.com/robotology/yarp/pull/1960#issuecomment-498317957
-  compiler: GCC 7

**Additional context**
Please note that binding generation in YCM 0.10.3 is broken anyway as described at https://github.com/robotology/ycm/issues/256, therefore YCM needs to be fixed and re-imported in YARP.

PeterBowman, 2019-06-04:
Maybe related, upon `make install`:

```
CMake Error at extern/ycm/cmake_install.cmake:79 (file):
  file INSTALL cannot find
  "/home/teo/repos/yarp/extern/ycm/ycm-0.10.3/cmake-3.8/Modules/CMakeParseArguments.cmake".
Call Stack (most recent call first):
  cmake_install.cmake:45 (include)
```

PeterBowman, 2019-06-07:
Still happening after https://github.com/robotology/yarp/pull/2047 (import YCM 0.10.4), bindings are unusable.

Tobias-Fischer, 2019-06-07:
See https://github.com/robotology/yarp/issues/1831

drdanz, 2019-06-10:
@PeterBowman Sorry, I cannot reproduce the install issue, anyway I'm closing this that was fixed by #2047, feel free to open a new issue if you still have problems.

closed 2019-06-10 by drdanz

PeterBowman, 2019-06-10:
Let me stress again this was **not** fixed by #2047 (and it has little to do with #1831), and IMO should be treated as a release blocker.