robotology/yarp #2038: imuBoschBNO055 has wrong conversion formulas for gyro and magnetometer

Nicogene, 2019-06-05:
In the `imuBoschBNO055` [implementation](https://github.com/robotology/yarp/blob/master/src/devices/imuBosch_BNO055/imuBosch_BNO055.cpp) there are these conversion formulas:
```c++
        // Fill in accel values
        data_tmp[3] = (double)raw_data[0] / 100.0;
        data_tmp[4] = (double)raw_data[1] / 100.0;
        data_tmp[5] = (double)raw_data[2] / 100.0;

        // Fill in Gyro values
        data_tmp[6] = (double)raw_data[6] / 100.0;
        data_tmp[7] = (double)raw_data[7] / 100.0;
        data_tmp[8] = (double)raw_data[8] / 100.0;

        // Fill in Magnetometer values
        data_tmp[9]  = (double)raw_data[3] / 100.0;
        data_tmp[10] = (double)raw_data[4] / 100.0;
        data_tmp[11] = (double)raw_data[5] / 100.0;
```
Looking in the Bosch documentation the unit measure are for **gyro**:
[image]

and for **magnetometer**:
[image]

Then it seems that the values have to be divided by 16.0 instead 100.0.

I ran a simple test moving the joint 0 of the head up to 20 deg @ 20 deg/s recording the gyro

# BEFORE (using 100 as scaling factor)
[image]
# AFTER (using 16 as scaling factor)
[image]

traversaro, 2019-06-05:
How do you know that the `AFTER` plot is correct?

traversaro, 2019-06-05:
cc @prashanthr05 @nunoguedelha

traversaro, 2019-06-05:
**This bug affects all the iCub robots containing an `imuBosch_BNO055`. In particular, the gyro and magnetometer data contained in `/icub/inertial` have been always not correct, for all YARP/icub-main releases.** 
The problem does not affect  older iCub that use an XSens IMU.

Nicogene, 2019-06-05:
Another confirmation using the `iKinGazeCtrl`:
# BEFORE
[image]

# AFTER 
[image]

traversaro, 2019-06-05:
There are a few issue that could be actually caused by this: 
* https://github.com/robotology/icub-support/issues?utf8=%E2%9C%93&q=iKinGazeCtrl

commit fd7687fe (2019-06-05): imuBosch: fix conversion factor for gyro and magnetometer

It fixes #2038
PR #2041 (2019-06-06): imuBosch: fix conversion factor for gyro and magnetometer
It fixes #2038
closed 2019-06-06 by drdanz