robotology/yarp #2043: Improvements to yarp::sig::Matrix function toString

kothiga, 2019-06-05:
**Is your feature request related to a problem? Please describe.**

I frequently use Yarp matrices because of their convenience, but recently I noticed that writing very large matrices to the network and using the built in _toString_ function takes a lot longer time than it should. Doing some investigating I found that the length of time for both these operations grew exponentially as the matrix grew.

Searching through the source files of Matrix I think I have found what causes this behaviour. On [line 185 of Matrix.cpp](https://github.com/robotology/yarp/blob/master/src/libYARP_sig/src/Matrix.cpp#L185) the string function _substr_ is utilised in a way that is rather inefficient. This method has a linear time complexity relative to the length of the string. Since the string that is getting put together calls the _substr_ method after each row, it ends up taking an exponentially longer amount of time as the size of the matrix grows.

**Describe the solution you'd like**

My suggested change is designed around completely eliminating the use of the String function _substr_ By putting a simple if statement dependent of the row/column iterator not being 0, we improve the performance of that Matrix function _toString_ drastically. An additional minor change to remove the redundant if/else statement based on the width would be it use a ternary operation inside of the string constructor.

    std::string Matrix::toString(int precision, int width, const char* endRowStr) const {

        //-- If the width is less than 1, use tabs, else use width number of spaces.
        std::string spacer((width<0) ? 1 : width, (width<0) ? '\t' : ' ');

        //-- Buffering.
        std::string ret = "";
        char buffer[350]; 
        const double* src = (*this).data();

        //-- Iterate through copying the contents from the matrix, into a string.
        //-- Avoid unnecessary string resizes by only adding spacers at the beginning.
        size_t r, c;
        for (r = 0; r < nrows; r++) {
             if (r) { ret += endRowStr; }
            for (c = 0; c < ncols; c++) {
                if (c) { ret += spacer; }
                sprintf(buffer, " %.*lf", precision, *src); src++;
                ret += buffer;
            }
        }

        return ret;
    }

Through preliminary testing I found my implementation to provide the exact same string as the current implementation in a fraction of the time.

**Describe alternatives you've considered**

My current work around to this time complexity issue has been to override the implementation with my own.

drdanz, 2019-06-06:
The comment to the function clearly states "Quick implementation, space for improvement.", therefore I'd say go for it! :smile:
It would be great if you could open a pull request...

kothiga, 2019-06-07:
That is good to hear! I will close this issue and open a pull request shortly. Thank you for the help.

closed 2019-06-07 by kothiga