robotology/yarp #2044: yarp_add_idl adds an extra `thrift` in the include folder

xEnVrE, 2019-06-06:
**Describe the bug**
 
I have a project to be built using `CMake` that uses `yarp_add_idl` to generate thrift `cpp` and `h` files.

Specifically, If `$ROOT` is the root of the sources  I have the thrift file in `$ROOT/thrift/<name>.thrift` containing some service of the form `service <Name>`.

Inside the `CMakeLists.txt`, I have some lines like
```
set(${EXE_TARGET_NAME}_THRIFT_HDR thrift/<name>.thrift)

yarp_add_idl(${EXE_TARGET_NAME}_THRIFT_SRC ${${EXE_TARGET_NAME}_THRIFT_HDR})

add_executable(${EXE_TARGET_NAME}
               ${${EXE_TARGET_NAME}_HDR}
               ${${EXE_TARGET_NAME}_SRC}
               ${${EXE_TARGET_NAME}_THRIFT_SRC}
)
```

Within my sources file, I am including the thrift header using `#include <thrift/<Name>.h>`

Starting from [this](https://github.com/robotology/yarp/commit/ebde91ff9bf31b013bd6e2dd39dcbbef4a9b7a59) commit, I cannot compile the project for the following reasons:
1. the header file `<Name>.h` within the build folder is placed in `build/src/<project-name>/include/thrift/thrift/` instead of  `build/src/<project-name>/include/thrift/` as happening in the past
1. the source file `<Name>.cpp`, that can be found in `build/src/<project-name>/src`, tries to include `thrift/<Name>.h` and this fails due to (1)
1. all my source tries to include  `thrift/<Name>.h` and this fails due to (1)

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04
 - yarp version: `3.1.0` ([this](https://github.com/robotology/yarp/commit/ebde91ff9bf31b013bd6e2dd39dcbbef4a9b7a59) commit)
-  compiler: `gcc 5.5.0`

drdanz, 2019-06-06:
Fixed in  b15ec0f

closed 2019-06-06 by drdanz

traversaro, 2019-10-23:
Obviously, this broke the code that was assuming that the `thrift` was indeed in the include folder, see https://github.com/robotology/wearables/issues/56 .

commit b15ec0fc (2019-06-06): YarpIDL: Fix output for nested thrift files

Fix #2044