robotology/yarp #2049: yarpdataplayer timstamp not consistent

lrapetti, 2019-06-10:
**Describe the bug**
The same dataset dumped with `yarpdatadumper` played multiple time seems to be played with different timing, and in particular this time seems to be related with the computational work that the machine is doing. 

**Expected behavior**
I would expect to be able to replay the same dataset with the real/same timestamp offline.

**Screenshots**
In particular I'm running the [this dataset](), and and the data are read from a consumer device ([human state provider](https://github.com/robotology/human-dynamics-estimation/tree/master/devices/HumanStateProvider)) reading the data from the port.
As you can see from the data, they were collected with both tx and rx time, so I have tried to play the data using one or the other, but the behaviour seems to be present.
Here you can some example of different duration by checking the `Time Taken` coloumn in the `yarpdataplayer` interface (I checked with a timer and that value seems coherent):
- `yarpdataplayer --withExtraTimeCol 1`
  - just playing the data
    [image]
  - data + consumer device
    [image]
- `yarpdataplayer --withExtraTimeCol 2`
  - just playing the data
    [image]
  - data + consumer device
    [image].

**Configuration (please complete the following information):**
 - OS: `MacOS 10.13.6`
 - yarp version: `3.1.1+10-20190514.4+git96bad8480`

@kouroshD

DanielePucci, 2019-07-20:
@drdanz @Nicogene may probably help us with this?

Nicogene, 2019-07-22:
It seems a clock drifting, I will give a look to this. 👍 
But maybe @vtikha , that is the author of `yarpdataplayer`, has some hints about what should be the problem here.

vtikha, 2019-07-22:
@Nicogene Yes, I had a f2f talk with @vvasco and I have started to look at it. It only happens with rxTime and not txTIme.

Nicogene, 2019-08-07:
Fixed by #2074

closed 2019-08-07 by Nicogene