robotology/yarp #2050: Unable to compile YARP_robottestingframework on Windows

Nicogene, 2019-06-20:
After renaming `RTF` in `robottestingframework` (see 06ae1329cfaa618c2af9457b8a4dee3a9cfb58ba, https://github.com/robotology/robot-testing-framework/issues/101, https://github.com/robotology/robot-testing-framework/pull/103) it is impossible to compile library containing the `robottestingframework` plugins a.k.a `YARP_robottestingframework` on windows, due to the reach the OS max path limit.
E.g.:
```
The expression "[MSBuild]::NormalizePath(C:\Users\ngenesio\robotology\robotology-superbuild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpplugin, robottestingframework_fixturemanager_yarpplugin.dir\Debug\, robottestingframework_fixturemanager_yarpplugin.vcxproj.CopyComplete)"
cannot be evaluated.
Path: C:\Users\ngenesio\robotology\robotology-superbuild\build\robotology\YARP\src\robottestingframework-plugins\fixture-managers\yarpplugin\robottestingframework_fixturemanager_yarpplugin.dir\Debug\robottestingframework_fixturemanager_yarpplugin.vcxproj.CopyComplete
exceeds the OS max path limit. The fully qualified file name must be less than 260 characters.  ALL_BUILD  C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets
```

Workaround(or fix):

https://www.howtogeek.com/266621/how-to-make-windows-10-accept-file-paths-over-260-characters/

Nicogene, 2019-06-20:
Actually only VS2019 support path longer than 260 characters, see: https://github.com/Microsoft/msbuild/issues/53#issuecomment-459062618

drdanz, 2019-06-20:
Not a YARP problem, you can change your build path to something shorter

closed 2019-06-20 by drdanz

traversaro, 2019-06-21:
@Nicogene feel free to open an issue on this in the superbuild, as it is contributing ~40 characters out of the max 260, and if we have a requirement  of minimum path length on Windows we should document it (similar to https://github.com/Microsoft/vcpkg/issues/426).