robotology/yarp #2052: yarpdataplayer speed not consistent with real data

vvasco, 2019-06-21:
**Describe the bug**
`yarpdataplayer` seems to reproduce data faster than the real data.
This seems related to #2049.

**To Reproduce**
- Record a dataset using `yarpdatadumper --rxTime --txTime`;
- Replay the dataset through `yarpdataplayer --withExtraTimeCol 2`;
- Record the same dataset from `yarpdataplayer` using `yarpdatadumper --rxTime --txTime`;
- Compare the receiver times.

**Expected behavior**
I expect `yarpdataplayer` to use the recorded times for playing the data, while it seems to reproduce data faster.

**Screenshots**
I recorded a dataset with the options `--rxTime --txTime` and re-dumped the data from the dataplayer with the same options. When plotting the receiver times of both datasets, it seems that `yarpdataplayer` (red line) reproduces the data faster than the real data (blue line), while I would expect times to be equal:

[image]

The delay accumulates and at the end of my dataset (which is ~`20mins` long), it reaches ~`158s`. 
`yarpdataplayer` seems to apply a warping factor of ~`0.8` to time, which seems consistent across different datasets. 

**Configuration (please complete the following information):**
 - OS: `Ubuntu 18.04.2 LTS`
 - yarp version: v3.1.0 commit `96123316f36edd00db68b46b2a6240d1d5ab11ba`

**Additional context**
I spotted the problem because I was trying to synchronize a command (black dashed line in the plot) using the recorded time, and it was given  ~`20s` later than expected.

pattacini, 2019-06-21:
Hi @vvasco 

Could you attach here in a zip containing the minimum dataset with which we can reproduce the problem?

vvasco, 2019-06-21:
Hi @pattacini 

Unfortunately I can't because the dataset contains sensitive data.
But I can provide @vtikha with it.

vtikha, 2019-08-05:
@vvasco & @lrapetti First of all I apologise for the delay but was not able to work on this any sooner. I managed to replicate the problem with your dataset and instructions. There was clearly an issue with the virtual virtual time of the `yarpdataplayer` and I have been working on a solution. 

**Behaviour**

I have used @vvasco 20 minutes dataset to:

- Replay the dataset through yarpdataplayer --withExtraTimeCol 2; (also --withExtraTimeCol 1)

- Record the same dataset from yarpdataplayer using yarpdatadumper --rxTime --txTime;

- Compare the receiver times (several times)

**Screenshot**
I plotted the receiver times of both datasets, it seems that the yarpdataplayer (red line) now manages to reproduce correctly the real data (blue line):

[image]

**Configuration:**

OS: MacOS 10.14.6
YARP version 3.2.0+20190710.4+git291925cd2

**steps**

I would like to do some more tests before creating a PR with the datasets from @lrapetti maybe. The updated code is now in the `fix-yarpdataplayer` branch.

Nicogene, 2019-08-07:
Fixed by #2074

closed 2019-08-07 by Nicogene