robotology/yarp #2054: Maps thrift idl msg not working properly

lrapetti, 2019-06-27:
I'm trying to use a thrift message defined using maps (the msg I am using is the following https://github.com/robotology/wearables/blob/feature/ICub-device-impl/msgs/thrift/WearableData.thrift), but I am experiencing the following problem after moving to `YARP 3.2.0` (where a major refactoring of idl thrift was done by @drdanz):
I can play the dataset with `yarpdataplayer` and with `yarp read ... /portname` I see the message beeing streamed correctly. However, when I try to access the port with a device implementing `TypedReaderCallback` (the I am using is: https://github.com/robotology/wearables/tree/feature/ICub-device-impl/devices/IWearRemapper), the `onRead()` method is called only once and the content of the message is empty. In fact:
- printing the message with the following line in `onRead()`
```
yInfo() << logPrefix << "ON READ: reading wearable data: " << receivedWearData.toString();
```
- it return the following:
```
[INFO]IWearRemapper : reading wearable data:  "XsensSuit::" () () () () () () () () () () () () () () () () 
```

After talking with @diegoferigo, he pointed out that the problem is analogous to what experienced in https://github.com/robotology/yarp/issues/1802, and may be a problem related to the deserialization.

drdanz, 2019-06-28:
@lrapetti Sorry, just to be sure, this worked before the upgrade to YARP 3.2, right?

lrapetti, 2019-06-28:
> Sorry, just to be sure, this worked before the upgrade to YARP 3.2, right?

@drdanz 
I have never tested it on the very same machine, but in a different machine I am at https://github.com/robotology/yarp/commit/96bad8480d657750c591970e310ba24f757e00b5 and it is working fine.

kouroshD, 2019-07-01:
Also, I have the same problem to get the information of data inside the thrift message.

traversaro, 2019-07-29:
Even without directly investigating the issue, writing a simple unit test on this in https://github.com/robotology/yarp/blob/master/tests/yarpidl_thrift/demo/main.cpp can be extremely helpful for expert YARP users to debug the issue and ensure no regression in the future.

lrapetti, 2019-07-30:
@drdanz is there any plan to handle this issue anytime soon? If needed I can help (e.g. what @traversaro suggested if you can give me some hints).

This issue is preventing us to upgrade to `YARP 3.2` most of the software and the laptop in the context of An.Dy project, so it would be good to try to tackle it in order to remain up-to-date with the latest release of YARP.

drdanz, 2019-08-01:
I'm sorry, I'm currently busy with other stuff, therefore I don't think I will be able to look at this before september...
If you want to try and fix it (it would be really appreciated), you can find the code of the generator in [src/idls/thrift/src/t_yarp_generator.cc](https://github.com/robotology/yarp/blob/master/src/idls/thrift/src/t_yarp_generator.cc)
You can enable some extra comment for debugging the generator by adding `debug_generator` to the command line.
I think you will have to check if the functions doing serialization and de-serialization are doing it properly... Perhaps, you can also compare the code generated from 3.1 with the one generated by 3.2, even if it is quite different, most changes don't impact the actual code of the functions...

Let me know if you decide to try and fix it yourself, and if you need help...

kouroshD, 2019-08-21:
@nunoguedelha

drdanz, 2019-09-17:
It would be really useful if you could give me a minimal thrift file to reproduce this issue

lrapetti, 2019-09-18:
I have developed a minimal example in which I am having problem moving from `YARP3.1` to `YARP3.2`.

The example can be found at https://github.com/lrapetti/test_thrift_map with the following content:
- two messages are defined, [with](https://github.com/lrapetti/test_thrift_map/blob/master/thrift/msgMap.thrift) and [without](https://github.com/lrapetti/test_thrift_map/blob/master/thrift/msg.thrift) `map`
- a [transmitter](https://github.com/lrapetti/test_thrift_map/blob/master/src/transmitter/transmitter.cpp) that streams both the messages in two different ports.
- a [receiver_no_map](https://github.com/lrapetti/test_thrift_map/blob/master/src/receiver/receiver_no_map.cpp) that read and print the content of the port without map
- a [receiver](https://github.com/lrapetti/test_thrift_map/blob/master/src/receiver/receiver_no_map.cpp) that read and print both the ports

What I experience is that in `YARP3.1` I am able to read both the ports with the `receiver`. With `YARP3.2` instead, the `receiver_no_map` works, while when I run the `receiver` the port stop streaming until I stop the receiver device.

drdanz, 2019-09-18:
Thanks.
I will work on this ASAP

drdanz, 2019-09-18:
From an initial debugging, the issue seem to be in the receiving part, the transmitted part is correct...

drdanz, 2019-09-18:
@lrapetti It should be fixed now, can you please try and report back?
Many thanks for the example, it was really helpful.
It would be great to integrate it in the unit tests...

closed 2019-09-18 by drdanz

drdanz, 2019-09-18:
> It would be great to integrate it in the unit tests...
Actually done in bfca0db72d5592e349aa93445459109a6a7df4ba :muscle:

lrapetti, 2019-09-19:
> @lrapetti It should be fixed now, can you please try and report back?

Both the transmitter/receiver example, and the thrift message mentioned in the [first comment](https://github.com/robotology/yarp/issues/2054#issue-461606699) seems to be working fine. Thanks a lot @drdanz!

GiulioRomualdi, 2019-09-20:
cc @xEnVrE

commit 235871e7 (2019-09-18): yarpidl_thrift: Fix deserialization for map types

Fixes #2054