robotology/yarp #2056: Vector initialization inside multipleanalogsensorsclient not working as expected.

fjandrad, 2019-07-08:
**Describe the bug**
When using the function getTemperatureSensorMeasure(size_t sens_index, double& out, double& timestamp) const
a dummy vector is created to store the results and give back the first value. This can be seen in  https://github.com/robotology/yarp/blob/48613af088ab5b3c41ddd6def6afaea0dd179b20/src/devices/multipleanalogsensorsclient/MultipleAnalogSensorsClient.cpp#L401

But when actually using this function the temperature value that we get is crazy.

**To Reproduce**
Steps to reproduce the behavior:
just call the function inserting a double in the out variable

**Expected behavior**
We would expect to get a reasonable measurement.

For the sake of trying I attempted to create a vector myself and then print the values of the vector, that gives me a reasonable measurement.  Here is the comparison of printing the results of both methods:
[image]

the first is when using a double the second when using a vector

**Configuration (please complete the following information):**
 - OS: linux
 - yarp version: YARP version 3.1.101+20190508.5+git98ffeb446
-  compiler: g++

valegagge, 2019-07-08:
Hi @fjandrad ,

when we spoke f2f, I understood you are using the MultipleAnalogSensorRemapper. I saw a bug here:
https://github.com/robotology/yarp/blob/3c157dc43783506b71fb03fe65ad9b201886aaaa/src/devices/multipleanalogsensorsremapper/MultipleAnalogSensorsRemapper.cpp#L509

The remapper creates `dummy`, a auxiliary vector of 1 element only, but then accesses to dummy[1], its second element.

Maybe your issue is due to this bug. 

Let me know if this issue persist.

fjandrad, 2019-07-08:
Tested and it solves the issue :+1:

Nicogene, 2019-08-06:
Closed by #2058

closed 2019-08-06 by Nicogene