robotology/yarp #2060: Invalid color conversion with mjpeg carrier

jchevrie, 2019-07-12:
**Describe the bug**

The `mjpeg` carrier does not perform the correct color conversion when connecting ports with different pixel formats.

**To Reproduce**

Connect a `BufferedPort<ImageOf <PixelBgr> >` to a `BufferedPort<ImageOf <PixelRgb> >` via the `mjpeg` carrier and send an image: the result is a received image with red and blue channels inverted.

**Expected behavior**

The correct color channels conversion should be performed to adapt to the pixel formats of the ports.

**Screenshots**

On the left a `BufferedPort<ImageOf<PixelBgr> >` sending an image (input image is red) to yarpview via `tcp` carrier, on the right the same port sending the same image via `mjpeg` carrier (should not be blue).

[image]

**Configuration**
 - OS: Ubuntu 18
 - yarp version: 3.2.0+20190710.3+git291925cd2
-  compiler: g++

jchevrie, 2019-07-12:
@Nicogene

Nicogene, 2019-07-15:
This piece of code allows to reproduce the bug:

``` c++
#include <yarp/os/BufferedPort.h>
#include <yarp/os/Network.h>
#include <yarp/os/LogStream.h>
#include <yarp/sig/Image.h>

using namespace yarp::sig;
using namespace yarp::os;

int main()
{
    Network yarp;
    BufferedPort<ImageOf<PixelBgr>> inPort;
    Port outPortBefore;
    if (!inPort.open("/testMjpeg:i") ||
        !outPortBefore.open("/testMjpeg") )
        return 1;
    if(!yarp.connect("/grabber", "/testMjpeg:i"))
        return 1;

    while(true)
    {
        Time::delay(0.1);
        auto inImg = inPort.read();

        if (inImg)
        {
            outPortBefore.write(*inImg);
        }
        else
        {
            return 1;
        }

    }
    return 0;
}
```

Nicogene, 2019-07-15:
The bug should be here:

https://github.com/robotology/yarp/blob/master/src/carriers/mjpeg_carrier/MjpegCarrier.cpp#L171L178

For now it supports only MONO or RGB

PR #2061 (2019-07-16): Fix mjpeg while using BGR images
This PR fixes #2060.

`MONO`, `RGB` and `BGR` have been successfully tested
commit 9043659d (2019-07-16): carriers: fix mjpeg carrier for BGR images

It fixes #2060
closed 2019-07-16 by drdanz