robotology/yarp #2068: Bump c++ requirement and supported compilers

drdanz, 2019-07-31:
According to [our policies](https://wiki.icub.org/wiki/YARP_Supported_Distributions), we can now drop CMake 3.5 on the devel branch.

For reference this is a list of c++ features supported by each compiler: https://en.cppreference.com/w/cpp/compiler_support

* I think we can drop support for visual studio 2015. We don't have an explicit policy for that, but I think that supporting the 2 most recent versions is enough. Visual Studio 2015 already supported C++14 with a couple of exceptions, Visual studio 2017 fully supports C++14. I assume that, whatever visual studio version people is using, they will update it to the latest update.
* Debian stretch/oldstable has the oldes gcc compiler (6.3.0) which is not what I would like to support, but still supports all the C++14

Therefore I don't see any problem in switching to c++14

* As soon as we drop support for stretch, the oldest compiler becomes 7.3.0. With a very limited subset of features that are not supported. This option is, in my opinion, to be taken really seriously, due to the **very important additions** to the language and to the standard library introduced by the standard.

So the options are

* Stay on **C++11**
* **C++14**
* **C++17** without the unsupported features

This is an important decision because it will impact the robots, and all the other repositories, therefore we need to discuss this.

cc @lornat75 @pattacini @vtikha @traversaro @Nicogene @aerydna

traversaro, 2019-07-31:
I think there are some missing words after 7.3.0 .

drdanz, 2019-08-01:
Fixed, thanks

drdanz, 2019-10-16:
The reason why this hasn't still be closed is because of the PC104.
I'm currently waiting to understand if there will be a new release of the image

drdanz, 2019-12-01:
The new PC104 image has been released.
Anyway, YARP 3.3 will require c++14 only, we will change it to c++17 after the release

PeterBowman, 2020-09-14:
Bug https://github.com/robotology/yarp/issues/1959 (apparently also affects Linux despite its current title) could take advantage of C++17's `if constexpr`.

closed 2020-12-22 by drdanz