robotology/yarp #2070: yarpmanager - cluster doesn't run /pc104 node

davidetome, 2019-08-01:
**Describe the bug**
running selected nodes in cluster tab they becomes green but in the application resource tab is in red.
If I run manually yarp run --server /pc104 it works

**Screenshots**
see attached screenshot
[image]

davidetome, 2019-08-01:
@pattacini @vtikha any suggestion? I should  fix it on the laptop of the iCubShenzen01 who's leaving the lab

cc @Fabrizio69

Nicogene, 2019-08-01:
>  I should fix it on the laptop of the iCubShenzhen01 who's leaving the lab

When it should be sent?

if it is urgent I can come and see what's wrong late this evening(after 17.30) or next week.

davidetome, 2019-08-01:
@Nicogene yes it's very urgent,  it should be sent yesterday!! 
if you come this late afternoon is ok; i can stay until 18.30/19.00.

closed 2019-08-01 by Nicogene

drdanz, 2019-08-01:
This issue was closed by @Nicogene, but was this fixed?

davidetome, 2019-08-01:
yes, thanks to @Nicogene  who fixed&closed

Nicogene, 2019-08-01:
> This issue was closed by @Nicogene, but was this fixed?

Yes fixed by 14946eb

commit 14946eba (2019-08-01): yarpmanager: fix cluster yarprun nodes

It fixes #2070