robotology/yarp #2071: yarpview / yarpscope crash (OpenGL)

arrenglover, 2019-08-02:
**Describe the bug**
On start-up of yarpview or yarpscope there is a segmentation fault. The GUI attempts to open, it flashes on the screen.

**To Reproduce**
> yarpserver
> yarpview

**Expected behavior**
yarpview/yarpscope should run and not crash.

**Screenshots**

> aglover@IITICUBLAP089:~$ yarpview
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-aglover' 
> yarp: Port /yarpview/img:i active at tcp://10.240.0.28:10002/                                                           
> QOpenGLShaderProgram: could not create shader program                                                                   
> QOpenGLShader: could not create shader                                                                                  
> QOpenGLShaderProgram::uniformLocation(imageTexture): shader program is not linked                                       
 > Segmentation fault (core dumped) 

**Configuration (please complete the following information):**
 - OS: 

Windows subsystem for linux
                                                                                   
> Distributor ID: Ubuntu                                                                                                  
> Description:    Ubuntu 18.04.2 LTS                                                                                      
> Release:        18.04                                                                                                   
> Codename:       bionic    

Windows 10 Pro

> Version	10.0.18362 Build 18362

 - yarp version:
> YARP Version: 3.2.100 (3.2.100+20190716.14+git8f7d73651)

-  compiler:
> --   CMake: 3.10.2 (/usr/bin/cmake)                                                                                     
> --   C Compiler: GNU 7.4.0 (/usr/bin/cc)                                                                                
> --   CXX Compiler: GNU 7.4.0 (/usr/bin/c++)                                                                             
> -- Build type: Release     

**Additional context**

It seems like it is a problem with my system rather than YARP itself. The problem only started happening on Wednesday (I might have done some system updates, I'm not sure). If I roll back yarp to master (release before Wednesday) the problem still occurs. Has anyone had similar problems recently or in the past? Google didn't help me fix it ...

Nicogene, 2019-08-02:
What are you using as X server? I encountered a similar problem(or maybe the same not sure) using [Xming](https://sourceforge.net/projects/xming/).

arrenglover, 2019-08-02:
Yes - sorry I forgot to write: yarpmanager GUI works fine

Yes I'm using xming (xming has been working for 2 years so far). I'll try a different xserver and see if it has the same behaviour

arrenglover, 2019-08-02:
@Nicogene thanks!

I tried vsxsrv instead of xming and it doesn't have the same problem. yarpview is working.

the latest version of vsxsrv seems to be doing everything that xming was doing so I can switch to it without any problems it seems. Thanks for pointing me in the right direction. If you want to close the issue go ahead.

Nicogene, 2019-08-02:
I'm happy that helped you 😄 

I would like to know if Microsoft(maybe with WSL2) has decided a "canonical" X server provider.

Otherwise it will be a mess to support WSL usage.

cc @traversaro @pattacini @claudiofantacci @drdanz

pattacini, 2019-08-02:
I can confirm that I've found in my experience `Xming` a bit buggy in this respect.
That's why I switched to `X410` in the past (from the Microsoft Store).

arrenglover, 2019-08-02:
okay good to know, but `X410` isn't free 😞 . `vsxsrv` had a new update in July that seems to be working nicely.

pattacini, 2019-08-02:
Yep not free, right @arrenglover 
They were offering it with a large discount some time ago so I caught the opportunity to get it for 10 dollars or similar.

arrenglover, 2019-08-05:
Update to this:
None of the X servers are actually working properly,
1. xming crashed with the above 
2. vcxsrv runs the app, and loads the window, but it is always blank/black. I get the following error:
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-aglover'                                           
> QGLXContext: Failed to create dummy context
3. x410 (trial) gets the following error and crashes before opening the window
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-aglover'                                           
> QGLXContext: Failed to create dummy context                                                                             
> yarp: Port /yarpview/img:i active at tcp://192.168.1.118:10003/                                                         
> Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options 
> QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize 
> -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior 
> QSurfaceFormat::SwapBehavior(DoubleBuffer), swapInterval 1, profile  
> QSurfaceFormat::OpenGLContextProfile(NoProfile))                                                                                                     
> Aborted (core dumped)

It seems possible to fix with a re-install of the graphics drivers, but I'm not sure exactly how that works WSL or for an X11 server. My guess is that some run-time library loaded by QLM related to the graphics is not the correct version. Any ideas?

pattacini, 2019-08-05:
`yarpscope` and `yarpview` are also crashing on my WSL installation: just tested right now.
Anyway, I can tell you that I say them working (at least `yarpview`) in the past.

randaz81, 2020-07-24:
Some updates on this question.

I have no problems running yarpview, etc on WSL. Even Gazebo runs super fast.
Here is my current configuration:

- Windows 10, Version 2004, Build 19041.329
- WSL1 or WSL2 (same for me)
- Nvidia Driver: 451.67
- VcXsrv (vesrion 64.1.20.8.1) configured with native opengl set to off.
- Inside WSL, the environment variable LIBGL_ALWAYS_INDIRECT=0 

_Please note that before updating the Nvidia Driver (it was <445.0) I had your same problem on WSL2._

closed 2020-07-24 by randaz81

arrenglover, 2020-07-24:
thanks @randaz81. I have recently tried with WSL again (1 & 2) and I am not having any problems either. It seems the problem has been fixed somewhere. 👍 I have basically the same set-up as you.