robotology/yarp #2077: Rewrite yarpview and yarpscope without QML

drdanz, 2019-08-14:
We never actually used any of the features offered by qml, but we had lots and lots of issues and bug reports due to the qml stuff.

Most of the issues regard:
* Missing menus
* Failed to find the qmldir or the qml plugin dll
* Filed to find runtime required plugins
* Issues with opengl

I think it's time that we give up on QML, and make them normal guis.

Opinions?

Nicogene, 2019-08-14:
It will solve probably the problems with the WSL and various Xservers(#2071)

I am not aware of the advantages using qml, but if we are not exploiting it, I agree to convert them and get rid of the qml-related troubles.

pattacini, 2019-08-18:
Agree on this 👍

traversaro, 2019-08-19:
This would also solve indirectly https://github.com/robotology/robotology-superbuild/issues/134 ?

closed 2020-12-22 by drdanz