robotology/yarp #2082: YARP_mathConfig.cmake wrongly formed

pattacini, 2019-08-18:
@drdanz the very last line of **`YARP_mathConfig.cmake` has no ending ")"**

```cmake
set(YARP_math_VERSION 3.2.100+20190816.176+git6a28a8d52)

####### Expanded from @PACKAGE_INIT@ by configure_package_config_file() #######
####### Any changes to this file will be overwritten by the next CMake run ####
####### The input file was YARP_mathConfig.cmake.in                            ########

get_filename_component(PACKAGE_PREFIX_DIR "${CMAKE_CURRENT_LIST_DIR}/../" ABSOLUTE)

####################################################################################

#### Expanded from @PACKAGE_DEPENDENCIES@ by install_basic_package_files() ####

include(CMakeFindDependencyMacro)
find_dependency(YARP_conf HINTS "${CMAKE_CURRENT_LIST_DIR}/.." NO_DEFAULT_PATH)
find_dependency(YARP_OS HINTS "${CMAKE_CURRENT_LIST_DIR}/.." NO_DEFAULT_PATH)
find_dependency(YARP_sig HINTS "${CMAKE_CURRENT_LIST_DIR}/.." NO_DEFAULT_PATH)

###############################################################################

include("${CMAKE_CURRENT_LIST_DIR}/YARP_mathTargets.cmake"
```

traversaro, 2019-08-20:
I think we have the same problem also for other components, this is a log coming from an internal CI system: 
~~~
-- Detecting CXX compile features - done
CMake Error: Error in cmake code at
<http://10.255.72.58:8080/job/Doxygen-***/ws/build/install/lib/cmake/YARP_OS/YARP_OSConfig.cmake>:34:
Parse error.  Function missing ending ")".  End of file reached.
CMake Error at <http://10.255.72.58:8080/job/Doxygen-***/ws/build/install/lib/cmake/YARP/YARPConfig.cmake>:184 (find_package):
  find_package Error reading CMake code from
  "<http://10.255.72.58:8080/job/Doxygen-***/ws/build/install/lib/cmake/YARP_OS/YARP_OSConfig.cmake".>
Call Stack (most recent call first):
  CMakeLists.txt:21 (find_package)
~~~

traversaro, 2019-08-20:
I am not able to reproduce the problem on Ubuntu 18.04 with CMake 3.10.2 , with external YCM  0.10.2 .

traversaro, 2019-08-20:
The issue is probably in a regression in YCM, as it happens with the current YCM `devel` branch, but not with YCM `master`.

traversaro, 2019-08-20:
The regression was introduced in https://github.com/robotology/ycm/pull/268, and a fix is proposed in https://github.com/robotology/ycm/pull/274 .

traversaro, 2019-08-21:
Fixed in https://github.com/robotology/ycm/pull/274 .

closed 2019-08-21 by traversaro