robotology/yarp #2083: Error when trying to use yarp::Thread in Python bindings

frank-foerster, 2019-09-02:
**Describe the bug**
I implemented a sub-class of yarp::Thread using YARP's Python bindings, but get a
`SWIG:DirectorMethodException` when executing the script. The precise error message is:

`terminate called after throwing an instance of 'Swig::DirectorMethodException'`
`what(): SWIG director method error. Error detected when calling 'Thread.threadInit'` 

**To Reproduce**
Create a sub-class of `yarp::Thread` using the Python bindings and try to instantiate it

**Expected behavior**
I would expect it to instantiate the Thread as a new object

**Configuration (please complete the following information):**
 - OS: Debian Linux
 - yarp version: 3.1.0+184
-  compiler: gcc 6.3.0

**Additional context**
I noticed that none of the very few python scripts in any of the robotology sub-repositories, that make use of multi-threading use YARP's own multi-threading functionality. All of them use native Python multi-threading libraries instead. This is what I am intending to do now as well but I thought it might be useful to solve this bug or, at least, understand somewhat better what causes it.

drdanz, 2019-09-05:
I think yarp threads should not be used with bindings, and the native implementation should be used instead. Nonetheless `yarp::Thread` should either work or not be created at all...
If it doesn't work, I vote for removing it.

jgvictores, 2020-01-09:
@frank-foerster Is it possible you ran into "Thread and RateThread need a Network yarp object" (https://github.com/robotology/yarp/issues/1595)?

I can replicate the error if the 5th line (`yarp.Network()`) is removed:
```python
import yarp
class MyThread(yarp.Thread):
    def run():
        print("hi")
yarp.Network() # or yarp.Network.init()
x = MyThread()
```

jgvictores, 2020-01-10:
Edited my above comment, was missing the `x.start()` line.

Unordered comments:
- Unfortunately, binding errors are hard to debug, `SWIG:DirectorMethodException` can appear even when issues are completely unrelated to SWIG.
- `RFModule` examples ([1](https://github.com/robotology/yarp/issues/1013#issue-195205722), [2](https://github.com/robotology/yarp/blob/1c6adb62a4b5f45bb33cf539b9f0363750f4560e/bindings/python/examples/example.py)) are still working (which is good).
- Long ago I played around inheriting from the `RateThread` class (now `PeriodicThread`). It worked, but things were laggy, so I ended up going for a more Pythonic solution.
- As a YARP Python binding user, nothing against hiding this specific component as proposed [above](https://github.com/robotology/yarp/issues/2083#issuecomment-530766782). As a general best practice, I guess we should favour idiomatic solutions, similar to using `time.sleep` rather than the YARPy equivalent.

drdanz, 2020-01-10:
After a couple of tests, I realized that this works:

```python
import yarp
class MyThread(yarp.Thread):
    def run(self):
        print("hi")
yarp.Network()
x = MyThread()
x.start()
```
 The only change is this one:

```diff
-    def run():
+    def run(self):
```

drdanz, 2020-01-10:
Probably relevant:

This works:

```python
    def run(self):
        yarp.delay(0.5)
```

This doesn't work:

```python
    def run(self):
        time.sleep(0.5)
```

```
terminate called after throwing an instance of 'Swig::DirectorMethodException'
  what():  SWIG director method error. Error detected when calling 'Thread.run'
```

closed 2020-07-14 by drdanz