robotology/yarp #2086: Clang error: move assignment operator in Searchable class

PeterBowman, 2019-09-04:
Originally reported at https://github.com/robotology/yarp/pull/2079#issuecomment-522273715. Affects YARP devel (v3.2.100, https://github.com/robotology/yarp/commit/639c4d39a2afa51fd22718d20e8a05971605f192).

```
/home/travis/YARP-devel/src/libYARP_OS/include/yarp/os/Searchable.h:103:17: error: 
      exception specification of explicitly defaulted move assignment operator
      does not match the calculated one
    Searchable& operator=(Searchable&& rhs) noexcept = default;
                ^
```

This error was spotted on a Travis CI job using Clang compiler; the corresponding GCC job succeeds. See logs at https://www.travis-ci.com/roboticslab-uc3m/tools/jobs/231222719.

```
OS: Ubuntu 16.04.6 LTS
clang version 7.0.0 (tags/RELEASE_700/final)
cmake version 3.12.4
````

drdanz, 2019-09-05:
I think this is a Clang bug, see https://bugs.llvm.org/show_bug.cgi?id=14141 (thanks @Nicogene for catching this)

drdanz, 2019-09-23:
I'm closing this, since I don't want to fix stuff for a specific version of a compiler that seems to be a compiler bug, please reopen if you have some reasons to believe it is not, or if you have a very good reason for wanting this fixed in YARP

closed 2019-09-23 by drdanz

PeterBowman, 2019-09-23:
I'll try to give my best in solving this. We configure all repos in our org to compile against GCC and Clang, hence every build on YARP devel fails due to this bug. I managed to successfully compile devel with Clang 6.0.0, so it's a pity Travis uses this exact Clang 7 buggy release (both Xenial and Bionic images, [ref](https://docs.travis-ci.com/user/languages/c/#clang-on-linux)).

> I think this is a Clang bug, see https://bugs.llvm.org/show_bug.cgi?id=14141

That's a different bug.

PeterBowman, 2019-10-07:
Possibly related: https://bugs.llvm.org/show_bug.cgi?id=38983 (not fixed yet).