robotology/yarp #2087: Yarp SerialDevice opening error

Yeshasvitvs, 2019-09-11:
I am trying to run the serialdevice on a **windows machine** as shown in the example https://github.com/robotology/yarp/tree/devel/src/devices/serialport

```
yarpdev.ext --device serialport --comport COM23 --file serial.ini

````

The serial port `COM23` and the contents of it are read using `Realterm` software and I can see the data streamed in the port. The file ` serial.ini` containts correct configuration parameters that are similar to those present in `Realterm` but still when I try to launch the yarp serial device, I get the following error:

```

[DEBUG]Subdevice serialport
[INFO]Starting Serial Port in COM23
[ERROR]Invalid communications port in COM23: No such file or directory
yarpdev: ***ERROR*** driver <serialport> was found but could not open
[ERROR]cannot make <serialport>
yarpdev: ***ERROR*** driver <serial> was found but could not open
[ERROR]yarpdev: ***ERROR*** device not available.
[INFO]Suggestions:
[INFO]+ Do "yarpdev --list" to see list of supported devices.
[INFO]+ Or append "--verbose" option to get more information.

```

On windows machine, `yarpdev.exe --list` does not work and the terminal is hung when tried to execute it. But alternatively, as @diegoferigo pointed out `yarp plugin --list`  shows `serialport`  in the list.

Yeshasvitvs, 2019-09-11:
@randaz81 please take a look at this issue. Thanks.

Yeshasvitvs, 2019-09-11:
I followed the tutorial in https://learn.sparkfun.com/tutorials/terminal-basics/command-line-windows-mac-linux and I can read the serial port data from the terminal. Yet, the yarp serial device is failing to start.

drdanz, 2019-09-13:
@randaz81 
* Is the "< COM9" requirement documented somewhere? If not perhaps it would be useful to write it down in the device documentation.
* It shouldn't be too difficult to add a check on the serial port name
* Is there a relative bug report/issue on ACE? If not we should open one.

closed 2020-07-07 by no-response[bot]