robotology/yarp #2089: Lazy symbol binding error on macOS 10.14.5

arlyon, 2019-09-16:
**Describe the bug**
- Install `yarp` according to the install guide [here](http://www.yarp.it/install_yarp_mac.html).
- Try to run it
- It crashes on launch with the following output:

```bash
    __  __ ___  ____   ____
    \ \/ //   ||  _ \ |  _ \
     \  // /| || |/ / | |/ /
     / // ___ ||  _ \ |  _/
    /_//_/  |_||_| \_\|_|
    ========================

Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
dyld: lazy symbol binding failed: Symbol not found: __ZN17ACE_SOCK_Acceptor4openERK8ACE_Addriiii
  Referenced from: /usr/local/Cellar/yarp/2.3.70.2/bin/../lib/libYARP_OS.1.dylib
  Expected in: /usr/local/opt/ace/lib/libACE.dylib

dyld: Symbol not found: __ZN17ACE_SOCK_Acceptor4openERK8ACE_Addriiii
  Referenced from: /usr/local/Cellar/yarp/2.3.70.2/bin/../lib/libYARP_OS.1.dylib
  Expected in: /usr/local/opt/ace/lib/libACE.dylib

[1]    64309 abort      yarpserver
```

**Configuration (please complete the following information):**
 - OS: `macOS 10.14.5`
 - yarp version: `robotology/formulae/yarp: stable 2.3.70.2 (bottled)`

I'm wondering if the contents of `libACE` has changed if it can't find the symbols it requires?

traversaro, 2019-09-16:
I am afraid that the YARP homebrew bottle was never updated for the change in ACE, and so it is broken.

traversaro, 2019-09-16:
Related issue: https://github.com/robotology/homebrew-formulae/issues/16 .

drdanz, 2019-12-03:
Closing this in favour of robotology/homebrew-formulae#16 .

closed 2019-12-03 by drdanz