robotology/yarp #2093: Make error

Urheen, 2019-09-20:
I am following the steps in http://www.yarp.it/install_yarp_linux.html#install_linux_from_sources 
Then when I type make, it showed two errors. I found a similar one in github, but the solution did not work for my  case. My log is below:

[ 74%] Built target yarpdev
Scanning dependencies of target yarprobotinterface
[ 74%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/main.cpp.o
[ 74%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Action.cpp.o
[ 74%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/CalibratorThread.cpp.o
[ 74%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Device.cpp.o
[ 74%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Module.cpp.o
[ 74%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Param.cpp.o
[ 75%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Robot.cpp.o
[ 75%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Types.cpp.o
[ 75%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/XMLReaderVx.cpp.o
[ 75%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/XMLReaderV1.cpp.o
[ 75%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/XMLReaderV3.cpp.o
[ 75%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/RobotInterfaceDTD.cpp.o
[ 75%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/idl_generated_code/src/yarprobotinterfaceRpc.cpp.o
[ 75%] Linking CXX executable ../../bin/yarprobotinterface
CMakeFiles/yarprobotinterface.dir/XMLReaderV1.cpp.o: In function `RobotInterface::XMLReaderFileV1::privateXMLReaderFileV1::readActionTag(TiXmlElement*)':
XMLReaderV1.cpp:(.text+0x5a56): undefined reference to `TiXmlAttributeSet::Find(std::string const&) const'
XMLReaderV1.cpp:(.text+0x5f6e): undefined reference to `TiXmlAttributeSet::Find(std::string const&) const'
CMakeFiles/yarprobotinterface.dir/XMLReaderV3.cpp.o: In function `RobotInterface::XMLReaderFileV3::privateXMLReaderFileV3::readActionTag(TiXmlElement*)':
XMLReaderV3.cpp:(.text+0x63b4): undefined reference to `TiXmlAttributeSet::Find(std::string const&) const'
XMLReaderV3.cpp:(.text+0x6adb): undefined reference to `TiXmlAttributeSet::Find(std::string const&) const'
collect2: error: ld returned 1 exit status
src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/build.make:268: recipe for target 'bin/yarprobotinterface' failed
make[2]: *** [bin/yarprobotinterface] Error 1
CMakeFiles/Makefile2:4309: recipe for target 'src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/all' failed
make[1]: *** [src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

traversaro, 2019-09-21:
Hi @Urheen , 
which  Linux distribution and of which version are you using?

Urheen, 2019-09-23:
Hi @traversaro 
I am using  Ubuntu 16.04 Xenial

Urheen, 2019-09-25:
Hi @traversaro, could you help me to solve this problem?

traversaro, 2019-09-25:
It seems to be a problem in the TinyXML version that you are using. Can you share the `CMakeCache.txt` file that you can find in the build directory of YARP?

Urheen, 2019-09-25:
Hi @traversaro , this is my CMakeCache.txt:

# This is the CMakeCache file.
# For build in directory: /home/ziywang/yarp/build
# It was generated by CMake: /opt/cmake-3.15.3-Linux-x86_64/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Path to a library.
ACE_ACE_LIBRARY_DEBUG:FILEPATH=ACE_ACE_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_ACE_LIBRARY_RELEASE:FILEPATH=/usr/lib/libACE.so

//Additional compiler flags for ACE
ACE_DEFINITIONS:STRING=

//Path to a library.
ACE_DL_LIBRARY_DEBUG:FILEPATH=ACE_DL_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_DL_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libdl.so

//directory containing ace/ACE.h for ACE library
ACE_INCLUDE_DIR:PATH=/usr/include

//ACE include directory
ACE_INCLUDE_DIRS:PATH=

//Path to a library.
ACE_RT_LIBRARY_DEBUG:FILEPATH=ACE_RT_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_RT_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/librt.so

//Allow YARP to (re)build IDL files as needed
ALLOW_IDL_GENERATION:BOOL=OFF

//path to the bison executable
BISON_EXECUTABLE:FILEPATH=BISON_EXECUTABLE-NOTFOUND

//Compile shared libraries rather than linking statically
BUILD_SHARED_LIBS:BOOL=ON

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Choose the type of build, options are: Release MinSizeRel Debug
// RelWithDebInfo DebugFull Profile.
CMAKE_BUILD_TYPE:STRING=Release

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//CXX compiler
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++

traversaro, 2019-09-26:
Hi @Urheen , this CMakeCache.txt seems truncated, are you sure that you copied all of it?

Urheen, 2019-09-27:
hi, @traversaro , sorry for the late reply and wrong information.
https://drive.google.com/file/d/1ODabOPI0kAFg5SoQGZoW6Mv-cvwyP1RI/view?usp=sharing
Above is the link for my CMakeCache.txt file.

Urheen, 2019-09-30:
hi, @traversaro, could you see the file I posted and help me to figure out the reason that I faced?

closed 2020-07-02 by drdanz

traversaro, 2020-07-02:
Sorry @Urheen, I had missed your messages. Did you fixed your problem in the end?

drdanz, 2020-07-02:
Sorry, I was going through the issues, and I closed a few that I thought were fixed, please feel free to reopen this if it is not