robotology/yarp #2094: Update the version in Ubuntu

Urheen, 2019-09-20:
when I install using sudo apt-get install icub, the version for icub is 1.12 while the latest is 1.13; and yarp is 3.1 while the latest is 3.2
Could you make the update in the Ubuntu so that we can directly install from it?

Urheen, 2019-09-30:
@mbrunettini, could you update the version in Ubuntu?

Urheen, 2019-10-08:
@mbrunettini @drdanz, do you have an estimate of when the updated version will be available for yarp? Thanks!

drdanz, 2019-10-09:
I'm not in charge of the packaging, therefore I cannot tell, sorry.

pattacini, 2019-10-09:
Hi @Urheen 

As we're currently considering a quite substantial overhaul of the way we provide binaries - for all platforms -, if you aim to get the latest releases, then our suggestion is to build the ecosystem from the source code, following the steps described in https://github.com/robotology/QA/issues/363.

To any rate, this does not rule out the possibility of still using yarp 3.1 and icub 1.12 from the available binaries.

We might update our wiki to reflect this ongoing transition.

Out of sheer curiosity, are you from a lab hosting an iCub?

Urheen, 2019-10-09:
Hi @pattacini 
Thank you for your reply. Currently I am facing the problems for directly installing the YARP 3.2 from source, and since the python-binding for icub 1.12 need  the Yarp with the version 3.2, so I am guessing whether I can get the update version of 3.2 from binaries.
Our lab does not host the physical robot, but in this project, we want to use the iCub simulator.

pattacini, 2019-10-10:
What kind of hitches did you come across while building YARP 3.2 from the source?

I'm not a Python user, but it sounds quite strange that the bindings in iCub 1.12 would rely on YARP 3.2. Can you elaborate more on this by detailing out the problem?

Urheen, 2019-10-15:
Hi @pattacini 
When I try to install python bindings for iCub, I got this error.
[image]
I believe this means that the python bindings for iCub rely on YARP 3.2.

Could you help me solve this problem?
Thank you very much

pattacini, 2019-10-15:
Hi @Urheen 

#### Let's try to clarify the context.
Latest `yarp` and `icub-main` binaries available on Linux are [`3.2.0`][1] and [`1.13.0`][2], respectively. You can check that from the code releases straight away. I also made an attempt to apt-install the binaries on a Ubuntu Bionic docker image and I got those exact versions. I did that since I usually build the ecosystem from the source code.

That ascertained, we can definitely close this issue.

Regarding your problem, my warm suggestion is to ask for a new support request on https://github.com/robotology/QA/issues, where you'll be required to provide a detailed description of your system (e.g. OS version) along with the steps you followed to recreate the ecosystem in terms of the source code version you downloaded (e.g. tags, branches, git hashes), the content of your environment variables, useful logs and so on so forth, that is everything that can help us narrow down this quirk.

All these aspects are unsaid or remain quite unclear from your posts above.

[1]: https://github.com/robotology/yarp/releases/tag/v3.2.0
[2]: https://github.com/robotology/icub-main/releases/tag/v1.13.0

closed 2019-10-16 by drdanz