robotology/yarp #2099: Compilation failed on Ubuntu 18.04 LTS

gabrielenava, 2019-09-24:
**Describe the bug**

The compilation on `master` branch fails on Ubuntu 18.04 with the following error:

```
/home/gcorsini/robotology-superbuild/robotology/YARP/src/libYARP_OS/src/Terminal.cpp:22:14: fatal error: editline/readline.h: No such file or directory
 #    include <editline/readline.h>
              ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
src/libYARP_OS/CMakeFiles/YARP_OS.dir/build.make:2798: recipe for target 'src/libYARP_OS/CMakeFiles/YARP_OS.dir/src/Terminal.cpp.o' failed
make[5]: *** [src/libYARP_OS/CMakeFiles/YARP_OS.dir/src/Terminal.cpp.o] Error 1
CMakeFiles/Makefile2:634: recipe for target 'src/libYARP_OS/CMakeFiles/YARP_OS.dir/all' failed
make[4]: *** [src/libYARP_OS/CMakeFiles/YARP_OS.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/YARP.dir/build.make:108: recipe for target 'robotology/YARP/CMakeFiles/YCMStamp/YARP-build' failed
make[2]: *** [robotology/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/YARP.dir/all' failed
make[1]: *** [CMakeFiles/YARP.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

```

I attempted at installing YARP through the robotology-superbuild on a new machine. I just installed all the system's dependencies following the [README](https://github.com/robotology/robotology-superbuild#linux) and then I attempted at compiling the repo.

**Configuration:**
 - OS: Ubuntu 18.04 LTS
 - yarp version: current `master` branch
-  compiler: cmake

cc @GiulioRomualdi was this the compilation problem you were mentioning recently?

traversaro, 2019-09-24:
I see two problems: 
* The robotology-superbuild does not list `libedit-dev` as a necessary apt package to install 
* For some reason, YARP does not complain at the CMake configuration level, but instead chokes during compilation. Do you tried with a clean build?

traversaro, 2019-09-24:
I see two problems: 
* `libedit-dev` should be listed in the apt dependencies of the robotology-superbuild 
* YARP should properly set its [`YARP_HAS_Libedit`](https://github.com/robotology/yarp/blob/eb0d831c886f10f714e65b70ec880c36a5a10210/src/libYARP_OS/src/Terminal.cpp#L21) variable based on system inspection, but for some reason this is not happening. Is this a clean build? Can you share the CMakeCache.txt of YARP?

Gianlucorso, 2019-09-24:
Hello,
@gabrielenava  tried to install robotology-superbuild on my machine.

Yes this should be a clean build.

In attachment you can find the 'CMakeCache' of Yarp.
[CMakeCache.txt](https://github.com/robotology/yarp/files/3647622/CMakeCache.txt)

traversaro, 2019-09-24:
It seems that on the computer there is a openrobots-distributed libedit, see: 
~~~
//Path to a library.
Libedit_EDIT_LIBRARY_RELEASE:FILEPATH=/home/gcorsini/openrobots/lib/libedit.so

//Libedit include directory
Libedit_INCLUDE_DIRS:PATH=/home/gcorsini/openrobots/include;/home/gcorsini/openrobots/include/editline
~~~

I think this is perfectly fine, but it highlighted that for some reason YARP build system is not propagating correctly the include dirs of libedit, and in the normal case they just work because the headers are installed in `/usr/include`.

traversaro, 2019-09-24:
Beside the strange YARP problem in using the Libedit headers, I guess installing a system version of libedit (`sudo apt install libedit-dev`) should solve your problem.

traversaro, 2019-09-24:
(OT, as this is not related to this YARP issue: did you updated to the latest superbuild by doing `git pull` and then `make update-all` in the build?

Gianlucorso, 2019-09-24:
I have installed [MATMPC](https://github.com/chenyutao36/MATMPC) which already contains a version of osqp and qpoases. 

I leave in attachment the `CMakeCache.txt` of OsqpEigen repository.  
[CMakeCache.txt](https://github.com/robotology/robotology-superbuild/files/3648073/CMakeCache.txt)

Gianlucorso, 2019-09-24:
I run `make VERBOSE=1` command only for `OsqpEigen` repository. Below you can find the result.

```
/usr/bin/cmake -H/home/gcorsini/robotology-superbuild/robotology/OsqpEigen -B/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen/CMakeFiles /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
make -f CMakeFiles/OsqpEigen.dir/build.make CMakeFiles/OsqpEigen.dir/depend
make[2]: Entering directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
cd /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/gcorsini/robotology-superbuild/robotology/OsqpEigen /home/gcorsini/robotology-superbuild/robotology/OsqpEigen /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen /home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen/CMakeFiles/OsqpEigen.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
make -f CMakeFiles/OsqpEigen.dir/build.make CMakeFiles/OsqpEigen.dir/build
make[2]: Entering directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
[ 25%] Building CXX object CMakeFiles/OsqpEigen.dir/src/Solver.cpp.o
/usr/bin/c++  -DOsqpEigen_EXPORTS -isystem /usr/include/eigen3 -I/home/gcorsini/robotology-superbuild/robotology/OsqpEigen/include -isystem /home/gcorsini/robotology-superbuild/build/install/include/osqp  -O3 -DNDEBUG -fPIC   -std=c++14 -o CMakeFiles/OsqpEigen.dir/src/Solver.cpp.o -c /home/gcorsini/robotology-superbuild/robotology/OsqpEigen/src/Solver.cpp
/home/gcorsini/robotology-superbuild/robotology/OsqpEigen/src/Solver.cpp: In member function ‘bool OsqpEigen::Solver::initSolver()’:
/home/gcorsini/robotology-superbuild/robotology/OsqpEigen/src/Solver.cpp:82:44: error: cannot convert ‘OSQPWorkspace**’ to ‘const OSQPData*’ for argument ‘1’ to ‘OSQPWorkspace* osqp_setup(const OSQPData*, OSQPSettings*)’
                   m_settings->getSettings()) != 0 ){
                                            ^
CMakeFiles/OsqpEigen.dir/build.make:110: recipe for target 'CMakeFiles/OsqpEigen.dir/src/Solver.cpp.o' failed
make[2]: *** [CMakeFiles/OsqpEigen.dir/src/Solver.cpp.o] Error 1
make[2]: Leaving directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/OsqpEigen.dir/all' failed
make[1]: *** [CMakeFiles/OsqpEigen.dir/all] Error 2
make[1]: Leaving directory '/home/gcorsini/robotology-superbuild/build/robotology/OsqpEigen'
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

gabrielenava, 2019-10-29:
problem has been solved and issue for OsqpEigen opened https://github.com/robotology/robotology-superbuild/issues/263. Closing the issue.

closed 2019-10-29 by gabrielenava