robotology/yarp #2101: Attribute Error: module 'yarp ' has no module named Network

ZiaMohiUDin, 2019-10-01:
**Describe the bug**
I have issue while controlling yarp by python.

**Expected behavior**
Icub controlled by python

**Configuration (please complete the following information):**
 - OS: windows 10
 - yarp 3.1.0 v14 
-  icub 1.12.0 v14 (installed from sources)

**Additional context**
yarpserver and icub are executing successfully. I cannot find yarp-so directory

drdanz, 2019-10-03:
This is definitely not enough information for helping you. Can you please share some code that I's causing you this issue, and the full log?

ZiaMohiUDin, 2019-10-03:
import collections
 import yarp
 
 yarp.Network.init() # Initialise YARP
 
 
 class WorldController:
     """Class for controlling iCub simulator via its RPC world port."""
     
     def __init__(self):
         self._rpc_client = yarp.RpcClient()
         self._port_name = "/WorldController-" + str(id(self)) + "/commands"
         self._rpc_client.open(self._port_name)
         self._rpc_client.addOutput("/icubSim/world")
         
         # A dictionary to track simulator object IDs for all types of objects
         self._sim_ids_counters = collections.defaultdict(lambda:0)
         
         # A sequence to track internal object IDs. This list stores tuples (object type, simulator id)
         # so that outside one does not have to remember the type of object.
         self._objects = [ ]
 
     def _execute(self, cmd):
         """Execute an RPC command, returning obtained answer bottle."""
         ans = yarp.Bottle()     
         self._rpc_client.write(cmd, ans)
         return ans
         
     def _is_success(self, ans):
         """Check if RPC call answer Bottle indicates successfull execution."""
         return ans.size() == 1 and ans.get(0).asVocab() == 27503 # Vocab for '[ok]'
         
     def _prepare_del_all_command(self):
         """Prepare the "world del all" command bottle."""
         result = yarp.Bottle()
         result.clear()
         map(result.addString, [ "world", "del", "all" ])
         return result
         
     def _prepare_create_command(self, obj, size, location, colour):
         """Prepare an RPC command for creating an object in the simulator environment.
         
         See Simulator Readme section 'Object Creation'
         
         Parameters:
             obj - object type string. 'sph', 'box', 'cyl' 'ssph', 'sbox' or 'scyl'.
             size - list of values specifying the size of an object. Parameters depend on object type:
                 (s)box: [ x, y, z ]
                 (s)sph: [ radius ]
                 (s)cyl: [ radius, length ]
             location - coordinates of the object location, [ x, y, z ]
             colour - object colour in RGB (normalised), [ r, g, b ]
         Returns:
             yarp.Bottle with the command, ready to be sent to the rpc port of the simulator
         
         """
         result = yarp.Bottle()
         result.clear()
     
         map(result.addString, ["world", "mk", obj])
         map(result.addDouble, size)
         map(result.addDouble, location)
         map(result.addDouble, colour)
         
         return result
 
     def _prepare_move_command(self, obj, obj_id, location):
         """Prepare the "world set <obj> <xyz>" command bottle."""
         result = yarp.Bottle()
         result.clear()
[... 58 lines cut ...]
         result = self._execute(self._prepare_get_command(obj_desc[0], obj_desc[1]))
         if result.size() == 3:
             return [ result.get(i).asDouble() for i in xrange(3) ] # 3-element list with xyz coordinates
         else:
             return None # An error occured
 
     def del_all(self):
         """Delete all objects from the simultor"""
         result = self._is_success(self._execute(self._prepare_del_all_command()))
         
         if result:
             # Clear the counters
             self._sim_ids_counters.clear()
             del self._objects[:]
         
         return result
 
     def __del__(self):
         try:
             if self._rpc_client <> None:
                 self.del_all()
             self._rpc_client.close()
             del self._rpc_client
         except AttributeError:
             pass

ZiaMohiUDin, 2019-10-03:
Thanks for your response,... I am looking forward

drdanz, 2019-10-04:
Thanks. Can you please post the execution log as well?

drdanz, 2019-10-07:
Please, next time instead of deleting the content of the issue content and change the title to some random characters just close the issue and add a comment that you solved it.

closed 2019-10-07 by drdanz

ZiaMohiUDin, 2019-10-07:
Okay sorry, I was unknown about that. I will take care next time.
Thanks for your correction