robotology/yarp #2103: OVR headset Does not work

GiulioRomualdi, 2019-10-07:
yarp device for ovr headset is not runnig.
We are using the yarp in `devel` and Oculus SDK version is `1.40.0`.
This the log file for the error:

[log_icub-virtualizer_yarpdev_7848.txt](https://github.com/robotology/yarp/files/3698309/log_icub-virtualizer_yarpdev_7848.txt)

GiulioRomualdi, 2019-10-07:
Looking at the code I discovered that the error
```
<ERROR> FrameTransformClient::getTransform() frames  mobile_base_body_link  and  head_link  are not connected 
```
is triggered only when the `relative` mode is enabled in the `ovrheadset` device.

In a previous version of yarp the `relative` mode was disabled.
https://github.com/robotology/yarp/commit/b2800f222352cd0cc410f6826269d432460b691a#diff-512556b5de879621a97ebd62f0575b8aL290

Now the `relative` flag is not set so the behavior of the code may be unpredictable.
What do you think about passing the `relative` parameter from configuration?

GiulioRomualdi, 2019-10-07:
This commit https://github.com/GiulioRomualdi/yarp/commit/aef216e33857c077d16cdd264510e192f439462c should fix the bug

drdanz, 2019-10-09:
@GiulioRomualdi Thanks for investigating this. Can you please open a pull request proposing these changes on `master`?

@aerydna Can you please have a look at the suggested changes?

aerydna, 2019-10-09:
ok for me!

drdanz, 2019-10-16:
Fixed by #2106

closed 2019-10-16 by drdanz