robotology/yarp #2116: Incompatibility between logger and exfat partition

bemilio, 2019-11-06:
**Describe the bug**
The LaCie 4TB d2 Professional hard drives that we have do not allow to copy folders that contain :o or :i at the end of the name. They are formatted with `exfat` filesystem. This conflicts with the names that are usually given to the yarp device ports and it renders more difficult to copy dumped datasets (since the folders are created with the ports names)

**To Reproduce**
Copy a dump of the robot data into one of the LaCie hard drives in the lab

**Expected behavior**
A correct copy

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04
 - yarp version:  3.2.1
-  compiler: g++

traversaro, 2019-11-06:
>  Copy a dump of the robot data into one of the LaCie hard drives in the lab

Which lab? : )

bemilio, 2019-11-06:
> Which lab? : )

The best lab :D 
DIC lab at CRIS building, IIT Genova

drdanz, 2019-11-13:
By `logger` you mean `yarpdatadumper`?

bemilio, 2019-11-13:
> By `logger` you mean `yarpdatadumper`?

yes

drdanz, 2019-11-14:
@bemilio If I understand correctly, the issue is that the `:` character is not allowed on exfat partition file names?

drdanz, 2019-11-14:
According to [wikipedia](https://en.wikipedia.org/wiki/ExFAT)

> Allowed characters in filenames | all Unicode characters except U+0000 (NUL) through U+001F (US) / (slash) \ (backslash) : (colon) * (asterisk) ? (question mark) " (quote) < (less than) > (greater than) and \| (pipe)(encoding in UTF-16LE)

bemilio, 2019-11-14:
> @bemilio If I understand correctly, the issue is that the `:` character is not allowed on exfat partition file names?

Exactly

drdanz, 2019-11-14:
Thanks.
For the dumping part, it should be a trivial issue to rename the files, but this might require also some extra work for the dataplayer side since, i believe, the port name is taken from the file name.
CC @vtikha

pattacini, 2019-11-14:
The port name is also reported plain within `info.log` as stated in the documentation: http://www.yarp.it/yarpdatadumper.html.

Don't know if `yarpdataplayer` reads that info though.

vtikha, 2019-11-14:
the ```yarpdataplayer``` get initialized with the the ```info.log``` file

pattacini, 2019-11-14:
I'm on it.

pattacini, 2019-11-14:
PR created (see #2122).

Anyway, `yarpdatadumper` can rely on the `--dir` command-line option to override dir names.

pattacini, 2019-11-14:
Done in #2122.

closed 2019-11-14 by pattacini