robotology/yarp #2117: yarp_portaudio does not compile on windows

GiulioRomualdi, 2019-11-08:
I installed `portaudio` using `.\vcpkg install portaudio --triplet x64-windows` on Windows10 (see https://github.com/microsoft/vcpkg/issues/8834 for further details).

When I tried to compile `yarp_portaudio` the building process returns the following error
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	C1083	Cannot open include file: 'portaudio.h': No such file or directory	yarp_portaudio	C:\Users\icub\sb\robotology\YARP\src\devices\portaudio\PortAudioDeviceDriver.h	27	
Error	C1083	Cannot open include file: 'portaudio.h': No such file or directory	yarp_portaudio	c:\users\icub\sb\robotology\yarp\src\devices\portaudio\PortAudioBuffer.h	23	
Error	C1083	Cannot open include file: 'portaudio.h': No such file or directory	yarp_portaudio	C:\Users\icub\sb\robotology\YARP\src\devices\portaudio\PortAudioDeviceDriver.h	27	
```

To understand what was going on i tried to check the content of `${PortAudio_LIBRARIES}`
https://github.com/robotology/yarp/blob/7ee03f8e042d31dcad271b6782dc1de8398a8b12/src/devices/portaudio/CMakeLists.txt#L30
and I noticed that it is empty.

Following the instruction of `vcpk` I decided to edit the `portaudio/CMakeLists.txt` file as follows 
```diff
diff --git a/src/devices/portaudio/CMakeLists.txt b/src/devices/portaudio/CMakeLists.txt index 49c5bf50b..356fb982a 100644
--- a/src/devices/portaudio/CMakeLists.txt
+++ b/src/devices/portaudio/CMakeLists.txt
@@ -26,8 +26,8 @@
-  target_include_directories(yarp_portaudio SYSTEM PRIVATE ${PortAudio_INCLUDE_DIR})
-  target_link_libraries(yarp_portaudio PRIVATE ${PortAudio_LIBRARIES}) 
+  target_link_libraries(yarp_portaudio PRIVATE portaudio portaudio_static))
```
This fixed the problem and now `yarp_portaudio` compiles on windows. However I don't know if this change break the compatibility with `linux` and `macOS`

@traversaro @vtikha

GiulioRomualdi, 2019-11-08:
The same considerations apply also for `yarp_portaudioPlayer` and `yarp_portaudioRecoreder`

Nicogene, 2019-11-08:
This change generates a linking issue on linux:

```bash
/usr/bin/ld: cannot find -lportaudio_static
clang: error: linker command failed with exit code 1 (use -v to see invocation)
src/devices/portaudio/CMakeFiles/yarp_portaudio.dir/build.make:149: recipe for target 'lib/yarp/yarp_portaudio.so' failed
make[2]: *** [lib/yarp/yarp_portaudio.so] Error 1
CMakeFiles/Makefile2:5414: recipe for target 'src/devices/portaudio/CMakeFiles/yarp_portaudio.dir/all' failed
make[1]: *** [src/devices/portaudio/CMakeFiles/yarp_portaudio.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

```

traversaro, 2019-11-08:
The problem is that vcpkg has the last (development :''( ) version of portaudio with proper imported targets, while Ubuntu/homebrew still use an older version of portaudio. Probably what it makes more sense is to modify the ycm's `FindPortAudio.cmake` script to support also the new PortAudio, and in particular in the case the new portaudio is detected, put the imported targets in `${PortAudio_LIBRARIES}`, so that we don't need to change anything in user code (in this case YARP).

Nicogene, 2019-11-11:
Fine for me, I'd move this issue to YCM then, it not seem a YARP problem

traversaro, 2020-10-14:
> 
> 
> The problem is that vcpkg has the last (development :''( ) version of portaudio with proper imported CMake targets, while Ubuntu/homebrew still use an older version of portaudio. Probably what it makes more sense is to modify the ycm's `FindPortAudio.cmake` script to support also the new PortAudio, and in particular in the case the new portaudio is detected, put the imported targets in `${PortAudio_LIBRARIES}`, so that we don't need to change anything in user code (in this case YARP).

Done in https://github.com/robotology/ycm/pull/357 .

drdanz, 2020-10-15:
If I understand correctly, the recent portaudio include a config file for the module `portaudio` (lowercase), but the new version is not yet in debian/ubuntu, it is not vcpkg adding the module. Am I right?

traversaro, 2020-10-15:
Exactly! The situation is that the recent portaudio from the master branch ships a portaudio CMake config, while the last released version from 2016 does not include it. vcpkg has it because it ships an unreleased version of portaudio.

drdanz, 2020-10-28:
I believe this can be closed now, see also https://github.com/robotology/robotology-superbuild/pull/499
Please reopen if you still have issues

closed 2020-10-28 by drdanz