robotology/yarp #2118: Failure finding thrifth headers in YARP3.3

lrapetti, 2019-11-11:
The PR #2102 broke back-compatibility with thrift files and include paths as described in https://github.com/robotology/yarp/pull/2102#issuecomment-546421417.

CC @drdanz

lrapetti, 2019-11-11:
More details and discussion can be found in https://github.com/robotology/wearables/issues/56.

drdanz, 2019-12-03:
Sorry, I cannot really understand what is happening here.
Can you produce a small test to help me understand the problem?

drdanz, 2019-12-03:
I think I was able to riproduce this, even if I'm not 100% sure that it is the same bug, wait for the test, I'll let you know if I still need it

drdanz, 2019-12-03:
I think the problem is related to having the `.thrift` files containing a `namespace` in a subdirectory
This leads to 4 possible positioning of the files:

* `<file>`
* `<namespace>/<file>`
* `<path>/<file>`
* `<path>/<namespace>/<file>`

This is handled by the thrift compiler with the `include_prefix` and `no_namespace_prefix` options that map to the `THRIFT_INCLUDE_PREFIX` and `THRIFT_NO_NAMESPACE_PREFIX` option in `yarp_idl_to_dir`
Unfortunately, `yarp_add_idl` does not allow these options and, for compatibility the `include_prefix` option is passed. I'm not completely sure, but I believe that the `no_namespace_prefix` should be passed as well for compatibility with YARP 3.1. For some reason that I cannot understand right now, I changed that in the patch.

I think the `<namespace>/<file>` is the one that makes more sense for `yarp_idl_to_dir`, but this means breaking compatibility, and therefore I didn't change it yet. Anyway, I should have reverted it to the old behaviour in #2136, and I'm going to defer this change to a future release, sorry if this will break compatibility later, I'll try to find a way to make it as smooth as possible.

drdanz, 2019-12-04:
Should be fixed in devel, please reopen this is it is not

closed 2019-12-04 by drdanz

traversaro, 2019-12-04:
I am afraid the are further regression due to this, see https://github.com/robotology/robotology-superbuild/runs/332895549 (not sure if it due to wearables or to YARP): 
~~~
2019-12-04T11:55:58.5536237Z FAILED: wrappers/IXsensMVNControl/CMakeFiles/IXsensMVNControlWrapper.dir/src/IXsensMVNControlWrapper.cpp.o 
2019-12-04T11:55:58.5537590Z /usr/bin/c++  -DIXsensMVNControlWrapper_EXPORTS -DUSING_DEPRECATED_UPPERCASE_YARP_OS_TARGET -I/home/runner/work/robotology-superbuild/robotology-superbuild/robotology/wearables/wrappers/IXsensMVNControl/include -Imsgs -I/home/runner/work/robotology-superbuild/robotology-superbuild/robotology/wearables/interfaces/IXsensMVNControl -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -g -fPIC   -std=gnu++14 -MD -MT wrappers/IXsensMVNControl/CMakeFiles/IXsensMVNControlWrapper.dir/src/IXsensMVNControlWrapper.cpp.o -MF wrappers/IXsensMVNControl/CMakeFiles/IXsensMVNControlWrapper.dir/src/IXsensMVNControlWrapper.cpp.o.d -o wrappers/IXsensMVNControl/CMakeFiles/IXsensMVNControlWrapper.dir/src/IXsensMVNControlWrapper.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/robotology/wearables/wrappers/IXsensMVNControl/src/IXsensMVNControlWrapper.cpp
2019-12-04T11:55:58.5538482Z /home/runner/work/robotology-superbuild/robotology-superbuild/robotology/wearables/wrappers/IXsensMVNControl/src/IXsensMVNControlWrapper.cpp:12:10: fatal error: XsensSuitControlService.h: No such file or directory
2019-12-04T11:55:58.5538999Z  #include "XsensSuitControlService.h"
2019-12-04T11:55:58.5539270Z           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
2019-12-04T11:55:58.5539532Z compilation terminated.
2019-12-04T11:55:58.5539895Z [50/53] Building CXX object wrappers/IXsensMVNControl/CMakeFiles/IXsensMVNControlWrapper.dir/yarp_plugin_ixsensmvncontrol_wrapper.cpp.o
2019-12-04T11:55:58.5540234Z [51/53] Building CXX object wrappers/IWear/CMakeFiles/IWearWrapper.dir/src/IWearWrapper.cpp.o
2019-12-04T11:55:58.5540510Z ninja: build stopped: subcommand failed. 
~~~

traversaro, 2019-12-04:
Yes, I can confirm that the same failure happens in a new build, and was not happening before https://github.com/robotology/yarp/pull/2136 was merged: 
https://github.com/robotology/robotology-superbuild/runs/332925188 

As this failure happens even with the workaround added in https://github.com/robotology/wearables/pull/65/files, I guess that technically it can be a different issue.  cc @lrapetti