robotology/yarp #2121: GrabberDual device doesn't work

valegagge, 2019-11-13:
**Describe the bug**
On purple iCub I tried to grab images from RGB camera (Leopard), but I get following error:
```
[WARNING]ServerGrabber: period parameter not found, using default of 0.03 s 
yarp: Port /icub/cam/rpc active at tcp://10.0.0.2:10003/
yarp: Port /icub/cam/left active at tcp://10.0.0.2:10004/
yarp: Port /icub/cam/right active at tcp://10.0.0.2:10005/
[WARNING]ServerGrabber: period parameter not found, using default of 0.03 s 
yarp: Port /icub/cam/rpc failed to activate at tcp://10.0.0.2:10003/ (address conflict)
[ERROR]ServerGrabber: unable to open rpc Port /icub/cam/rpc 
yarp: Port /icub/cam/left failed to activate at tcp://10.0.0.2:10004/ (address conflict)
[ERROR]ServerGrabber: unable to open image streaming Port /icub/cam/left 
yarp: Port /icub/cam/right failed to activate at tcp://10.0.0.2:10005/ (address conflict)
[ERROR]ServerGrabber: unable to open image streaming Port /icub/cam/right 
[ERROR]ServerGrabber: Error initializing YARP ports 
yarpdev: ***ERROR*** driver <grabberDual> was found but could not open
[ERROR]opening  subdevice... FAILED
[ERROR]ServerGrabber: error while opening subdevice
yarpdev: ***ERROR*** driver <grabberDual> was found but could not open
[ERROR]yarpdev: ***ERROR*** device not available.
[INFO]Suggestions:
[INFO]+ Do "yarpdev --list" to see list of supported devices.
[INFO]+ Or append "--verbose" option to get more information.
```

**To Reproduce**
In a clean setup (no yarp application is running)
1. launch `yarprobotinterface`
1. launch `yarpdev --from camera/ServerGrabberDualLeopard.ini` (obviously on icub-head)

**Expected behavior**
Obviously I expect to grab images. Moreover the above command is used in some application of icub.

**Additional context**
I briefly tried to get into the issue and it seems to me that that the `open` function of the device is call more than one time. In fact, the `address conflict` indicates that the the device is already opened like the reported log indicates.
I managed to grab image only using ` usb camera device` directly, instead of `grabberdual` device and `usbcamera` like subdevice.

Nicogene, 2019-11-18:
It is an bug that I saw some time ago, I hoped that it fixed by itself :sweat_smile: 

I'll look into it :eyes:

Nicogene, 2019-12-11:
Fixed by https://github.com/robotology/yarp/commit/a33127dc8a0ff4478b2e97145d3c59e671f70ccb

closed 2019-12-11 by Nicogene

PeterBowman, 2020-01-11:
"Fixed in: YARP yarp-3.3" label -> fixed in YARP master (3.4) instead?

Nicogene, 2020-01-14:
> "Fixed in: YARP yarp-3.3" label -> fixed in YARP master (3.4) instead?

It should be ok `yarp-3.3`, when `3.3.1` will be released it will become `Fixed in : YARP v3.3.1`

PeterBowman, 2020-01-14:
My bad, I thought commit https://github.com/robotology/yarp/commit/a33127dc8a0ff4478b2e97145d3c59e671f70ccb was not available on the yarp-3.3 branch, just master (see reference to branch below the commit title and description). Sorry!

commit a33127dc (2019-12-11): grabberDual: fix double open

It fixes #2121