robotology/yarp #2125: Issues with libACE on recent distros

xEnVrE, 2019-11-20:
**Describe the bug**
This is not strictly related to `YARP`, however, since `YARP` depends on `libace`, I think it is better to report it.

It seems that some distros, e.g. `Fedora  >= 9`, `Ubuntu >= 19.10`, `Arch Linux` (and many others I am not ware of) are removing the file `/usr/include/stropts.h` that is included within libace. For Linux, the developers of libace are handling this in the configuration file `$ACE_SOURCE_ROOT/ACE_WRAPPERS/ace/config-linux.h` with something like this

```
// Starting with FC9 rawhide this file is not available anymore but             
// this define is set
#if defined _XOPEN_STREAMS && _XOPEN_STREAMS == -1                              
# define ACE_LACKS_STROPTS_H                                                    
# define ACE_LACKS_STRRECVFD                                                    
#endif 
```
as `libace` can work even without the support to these streams.

However, this works only on `Fedora` as far as I know. For example, on `Arch Linux` this does not work, ditto for  `Ubuntu >= 19.10` as @marcoaccame reported.

Of course, this means that `libace` does not compile on these systems. Also, it seems that the package `libace-dev` available on `Ubuntu 19.10` is shipped with header files that include `stropts.h` (probably because the package is built with support to stropts enabled) resulting in a failing build for `YARP` as soon as `/usr/include/ace/os_include/os_stropts.h` is included.

**Configuration (please complete the following information):**
 - OS: any Linux distro missing `/usr/include/stropts.h`
 - libace version: I tested with `6.4.8` while @marcoaccame uses `6.4.5` shipped with `Ubuntu 19.10`

Nicogene, 2019-11-20:
Yet another reason to push https://github.com/robotology/yarp/issues/1637.
Unfortunately the 20.04 release it is too close in order to get rid of ACE

traversaro, 2019-11-20:
As a workaround, can't we just add `ACE_LACKS_STROPTS_H` to the YARP compile definitions? If I understand correctly, this should avoid  inclusion of the missing header `stropts.h` . 

By the way, I think this is actually connected to this bug in Gazebo : https://bugs.launchpad.net/ubuntu/+source/ace/+bug/1852425 .

traversaro, 2019-11-20:
@xEnVrE @marcoaccame  I think a possible workaround for this is to add a 
~~~
if(TARGET ACE::ACE)
  target_compile_definitions(ACE::ACE INTERFACE ACE_LACKS_STROPTS_H) 
endif()
~~~
after the `find_package(ACE ...)` in https://github.com/robotology/yarp/blob/218aff62ea404fdb0001e28dbcdc78cdf456487b/cmake/YarpFindDependencies.cmake#L275 ?

traversaro, 2019-11-20:
There was also a ACE upstream related fix : https://github.com/DOCGroup/ACE_TAO/pull/939 .

traversaro, 2019-11-20:
I updated the issue title as the issue is related to distro with glibc >= 2.30, see https://savannah.gnu.org/forum/forum.php?forum_id=9515 .

davidetome, 2019-11-29:
> @xEnVrE @marcoaccame I think a possible workaround for this is to add a
>
> ```
> if(TARGET ACE::ACE)
>   target_compile_definitions(ACE::ACE INTERFACE ACE_LACKS_STROPTS_H)
> endif()
> ```
>
> after the [...]

@traversaro , please see https://github.com/robotology/yarp/issues/2128#issuecomment-559494285

drdanz, 2019-11-29:
Can anyone with glib 2.30 try if something like this works?

```diff
diff --git a/extern/ycm/ycm-0.10.4/find-modules/FindACE.cmake b/extern/ycm/ycm-0.10.4/find-modules/FindACE.cmake
index 2deda332c..f1d2515fc 100644
--- a/extern/ycm/ycm-0.11.0/find-modules/FindACE.cmake
+++ b/extern/ycm/ycm-0.11.0/find-modules/FindACE.cmake
@@ -94,6 +94,14 @@ if (NOT DEFINED ACE_INLINE OR ACE_INLINE)
   set(ACE_DEFINITIONS "__ACE_INLINE__")
 endif()
 
+if(UNIX AND NOT APPLE)
+  include(CheckIncludeFileCXX)
+  check_include_file_cxx(stropts.h ACE_HAS_STROPTS_H)
+  if(NOT ACE_HAS_STROPTS_H)
+    list(APPEND ACE_DEFINITIONS ACE_LACKS_STROPTS_H)
+  endif()
+endif()
+
 ########################################################################
 ## OS-specific extra linkage
 
@@ -172,9 +180,7 @@ add_library(ACE::ACE UNKNOWN IMPORTED)
 set_target_properties(ACE::ACE PROPERTIES INTERFACE_INCLUDE_DIRECTORIES "${ACE_INCLUDE_DIRS}")
 set_target_properties(ACE::ACE PROPERTIES INTERFACE_SYSTEM_INCLUDE_DIRECTORIES "${ACE_INCLUDE_DIRS}")
 set_target_properties(ACE::ACE PROPERTIES INTERFACE_LINK_LIBRARIES "${ACE_EXTRA_LIBRARIES}")
-if (NOT DEFINED ACE_INLINE OR ACE_INLINE)
-  set_target_properties(ACE::ACE PROPERTIES INTERFACE_COMPILE_DEFINITIONS "__ACE_INLINE__")
-endif()
+set_target_properties(ACE::ACE PROPERTIES INTERFACE_COMPILE_DEFINITIONS "${ACE_DEFINITIONS}")
 
 if(ACE_ACE_LIBRARY_RELEASE)
   set_property(TARGET ACE::ACE APPEND PROPERTY IMPORTED_CONFIGURATIONS RELEASE)
```

traversaro, 2019-11-29:
> > @xEnVrE @marcoaccame I think a possible workaround for this is to add a
> > ```
> > if(TARGET ACE::ACE)
> >   target_compile_definitions(ACE::ACE INTERFACE ACE_LACKS_STROPTS_H) 
> > endif()
> > ```
> > 
> > 
> > after the `find_package(ACE ...)` in
> > https://github.com/robotology/yarp/blob/218aff62ea404fdb0001e28dbcdc78cdf456487b/cmake/YarpFindDependencies.cmake#L275
> > 
> > ?
> 
> @traversaro , please see [#2128 (comment)](https://github.com/robotology/yarp/issues/2128#issuecomment-559494285)

I did not consider the CMake version in my solution. The one suggested  by @drdanz in https://github.com/robotology/yarp/issues/2125#issuecomment-559741190 should work fine instead.

drdanz, 2019-11-29:
After some considerations, and some debugging with @xEnVrE, I believe there is nothing that we can do at our side, the bug was fixed upstream in ACE, (see https://github.com/DOCGroup/ACE_TAO/commit/e7e38f991ff56b3c628c1cb0b891be8a4c125b1d) and is available in ACE 6.5.7.

I suggest to report a bug in your distros (arch/ubuntu), and ask them to update the ACE package

drdanz, 2019-11-29:
A quick and dirty fix is to change this in `/usr/include/ace/config-linux.h` installed by the `libace-dev` package

```diff
-#if defined _XOPEN_STREAMS && _XOPEN_STREAMS == -1
+#if !defined(_XOPEN_STREAMS) || (defined _XOPEN_STREAMS && _XOPEN_STREAMS == -1)
```

closed 2019-11-29 by drdanz

drdanz, 2019-11-29:
Or perhaps also adding `-D_XOPEN_STREAMS=-1` should work

triccyx, 2020-01-31:
I can confirm the workaround on ubuntu 19.10 workshttps://github.com/robotology/yarp/issues/2125#issuecomment-559786737

drdanz, 2020-03-19:
For reference, glibc 2.30 just entered debian testing.
I just reported a bug: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=954252

drdanz, 2020-04-17:
It should now be fixed in debian `libace-dev` 6.4.5+dfsg-1.1, see

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=946853

drdanz, 2020-04-17:
Sync in ubuntu requested: https://bugs.launchpad.net/ubuntu/+source/ace/+bug/1873445