robotology/yarp #2126: Yarpdataplayer speed control not working

xEnVrE, 2019-11-23:
**Describe the bug**
`yarpdataplayer` speed control is not working.

**To Reproduce**
Open any recorder dataset with `yarpdataplayer` and tries to change speed using the slider in the GUI.

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04
 - yarp version: latest devel
-  compiler: g++ (Ubuntu 7.4.0-1ubuntu1~18.04.1)

**Additional context**

It seems that the variable `utilities->speed` is set in the GUI code

https://github.com/robotology/yarp/blob/632996c534a7cbd7bac3690e1cf23423f5bab166/src/yarpdataplayer/src/mainwindow.cpp#L901-L914

but it is not used elsewhere

```
$: yarp/src/yarpdataplayer$ grep -r "speed" .
./src/mainwindow.cpp:    QString speed = QString("%1").arg(szValue);
./src/mainwindow.cpp:    ui->speedValueLbl->setText(speed);
./src/mainwindow.cpp:        utilities->speed = value;
./src/utils.cpp:    speed(1.0),
./include/utils.h:    double              speed;
./include/UI/mainwindow.ui:          <widget class="QLabel" name="speedLbl">
./include/UI/mainwindow.ui:          <widget class="QLabel" name="speedValueLbl">

```

vtikha, 2019-11-23:
on it...

vtikha, 2019-11-23:
This has been handled in #2127. Thanks @xEnVrE

Nicogene, 2019-11-29:
Fixed by #2127

closed 2019-11-29 by Nicogene