robotology/yarp #2128: compilation failure

davidetome, 2019-11-28:
just cloned on a clean VM (pop_os - ubuntu 19.10), on master branch. See picture below
[image]

cc @mbrunettini

traversaro, 2019-11-28:
Hi @davidetome, the issue has been already reported in https://github.com/robotology/yarp/issues/2125 . There you can also find a possible solution/workaround in https://github.com/robotology/yarp/issues/2125#issuecomment-556036924 , if you could test it it would be great (I do not have any Ubuntu 19.10 to quickly test the problem).

traversaro, 2019-11-28:
If it is ok for you, I would close this issue as a duplicate of https://github.com/robotology/yarp/issues/2125 .

closed 2019-11-28 by Nicogene

davidetome, 2019-11-28:
> Hi @davidetome, the issue has been already reported in #2125 . There you can also find a possible solution/workaround in [#2125 (comment)](https://github.com/robotology/yarp/issues/2125#issuecomment-556036924) , if you could test it it would be great (I do not have any Ubuntu 19.10 to quickly test the problem).

added **if statement** at line 276; i get the following:

[image]