robotology/yarp #2133: PR https://github.com/robotology/yarp/pull/2129 breaks find_package(YARP) on Windows

traversaro, 2019-12-01:
**Describe the bug**
After https://github.com/robotology/yarp/pull/2129, `find_package(YARP)`does not work anymore, at least when YARP is using an external YCM 0.10.4 .

**To Reproduce**
Steps to reproduce the behavior:
Configure the CMake build of a project that depends on YARP, such as ICUB on Windows. It will fail with the following error: 
~~~
2019-11-30T02:22:33.1476528Z   -- Could NOT find YARP_os (missing: YARP_os_DIR)
2019-11-30T02:22:33.1485668Z   -- Could NOT find YARP_sig (missing: YARP_sig_DIR)
2019-11-30T02:22:33.1496019Z   -- Could NOT find YARP_dev (missing: YARP_dev_DIR)
2019-11-30T02:22:33.1505703Z   -- Could NOT find YARP_math (missing: YARP_math_DIR)
2019-11-30T02:22:33.1515021Z   -- Could NOT find YARP_idl_tools (missing: YARP_idl_tools_DIR)
2019-11-30T02:22:33.1635013Z   -- Found YARP: D:/a/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.2.102+20191129.16+git93a1418df")
2019-11-30T02:22:33.1638914Z   -- YARP is version: 3.2.102+20191129.16+git93a1418df
2019-11-30T02:22:33.1639577Z   CMake Error at CMakeLists.txt:23 (message):
2019-11-30T02:22:33.1640311Z     YARP was found, but no libYARP_math was detected, please recompile yarp
2019-11-30T02:22:33.1640688Z   
2019-11-30T02:22:33.1665871Z   -- Configuring incomplete, errors occurred!
~~~

**Expected behavior**
All the components of YARP should be found correctly. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Configuration (please complete the following information):**
 - OS: Windows
 - yarp version: https://github.com/robotology/yarp/commit/ee008e6a5f95cf4d15c4b1c0e244b211c6531149
-  compiler: Visual Studio 2019

**Additional context**
We detected the problem in robotology-superbuild CI, in particular: 
* https://github.com/robotology/robotology-superbuild/commit/1bc96a07b26484e0c1af386997cb319f734da082/checks?check_suite_id=335425431

pattacini, 2019-12-01:
I can confirm this is happening only when YARP is installed. Instead, when it is used from the build, it gets found correctly.

traversaro, 2019-12-01:
> I can confirm this is happening only when YARP is installed. Instead, when it is used from the build, it gets found correctly.

Are you using an internal YCM or an external one? In the case of external YCM, which version?

traversaro, 2019-12-01:
Apparently the failure  is present also when using YCM 0.11,  see https://github.com/robotology/robotology-superbuild/commit/39a6ba91209023b65cde7260097d159b86adead1/checks .

pattacini, 2019-12-01:
>Are you using an internal YCM or an external one? In the case of external YCM, which version?

Using `YCM/devel`.

drdanz, 2019-12-03:
I think that the problem is that I updated the path in YarpDescribe.cmake to install in `lib/cmake/YARP`, but the `InstallBasicPackageFiles` module still installs in the `CMake` folder.

So, if I'm correct, after the installation there should be a file `<install_prefix>/lib/cmake/YARP/YARPConfig.cmake` and a file `<install_prefix>/CMake/YARP_os/YARP_osConfig.cmake`. Can anyone confirm this?

Passing `INSTALL_DESTINATION` should fix this issue...

Can anyone on windows try this? https://github.com/drdanz/yarp/commit/8a473bedc3363a4177ff35e64400fe8ced11cb64

traversaro, 2019-12-03:
> Can anyone on windows try this? [drdanz@8a473be](https://github.com/drdanz/yarp/commit/8a473bedc3363a4177ff35e64400fe8ced11cb64)

CI job for testing this started in https://github.com/robotology/robotology-superbuild/pull/309 .

traversaro, 2019-12-03:
> > Can anyone on windows try this? [drdanz@8a473be](https://github.com/drdanz/yarp/commit/8a473bedc3363a4177ff35e64400fe8ced11cb64)
> 
> CI job for testing this started in [robotology/robotology-superbuild#309](https://github.com/robotology/robotology-superbuild/pull/309) .

Ah, I did not noticed it was on your fork.

traversaro, 2019-12-03:
> Ah, I did not noticed it was on your fork.

Fixed in https://github.com/robotology/robotology-superbuild/pull/309/files#diff-6943a71020735fd742ba0f16bf95afb5 .

drdanz, 2019-12-03:
Thanks @traversaro
I will commit it on devel as soon as the build is successful

drdanz, 2019-12-03:
The windows build were successful, but the macos builds failed. Do you think this is related?
Anyway, I'm going to push the change, since this issue seem to be fixed

closed 2019-12-03 by drdanz

traversaro, 2019-12-03:
> The windows build were successful, but the macos builds failed. Do you think this is related?

Not sure, I could not find anything related to the failure in the build log. I keep you posted if the problem results to be reproducible.

commit 95d3be06 (2019-12-03): CMake: Install all CMake files in lib/cmake/YARP_xxx also for windows

Fixes #2133