robotology/yarp #2142: Regression in yarp_idl_to_dir behaviour in YARP 3.3.0 w.r.t 3.2

traversaro, 2019-12-05:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Clone the repo https://github.com/robotology/wearables (that depends just on YARP), configure and build it.
The compilation fails with:
~~~
2019-12-04T11:55:58.5536237Z FAILED: wrappers/IXsensMVNControl/CMakeFiles/IXsensMVNControlWrapper.dir/src/IXsensMVNControlWrapper.cpp.o 
2019-12-04T11:55:58.5537590Z /usr/bin/c++  -DIXsensMVNControlWrapper_EXPORTS -DUSING_DEPRECATED_UPPERCASE_YARP_OS_TARGET -I/home/runner/work/robotology-superbuild/robotology-superbuild/robotology/wearables/wrappers/IXsensMVNControl/include -Imsgs -I/home/runner/work/robotology-superbuild/robotology-superbuild/robotology/wearables/interfaces/IXsensMVNControl -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -g -fPIC   -std=gnu++14 -MD -MT wrappers/IXsensMVNControl/CMakeFiles/IXsensMVNControlWrapper.dir/src/IXsensMVNControlWrapper.cpp.o -MF wrappers/IXsensMVNControl/CMakeFiles/IXsensMVNControlWrapper.dir/src/IXsensMVNControlWrapper.cpp.o.d -o wrappers/IXsensMVNControl/CMakeFiles/IXsensMVNControlWrapper.dir/src/IXsensMVNControlWrapper.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/robotology/wearables/wrappers/IXsensMVNControl/src/IXsensMVNControlWrapper.cpp
2019-12-04T11:55:58.5538482Z /home/runner/work/robotology-superbuild/robotology-superbuild/robotology/wearables/wrappers/IXsensMVNControl/src/IXsensMVNControlWrapper.cpp:12:10: fatal error: XsensSuitControlService.h: No such file or directory
2019-12-04T11:55:58.5538999Z  #include "XsensSuitControlService.h"
2019-12-04T11:55:58.5539270Z           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
2019-12-04T11:55:58.5539532Z compilation terminated.
2019-12-04T11:55:58.5539895Z [50/53] Building CXX object wrappers/IXsensMVNControl/CMakeFiles/IXsensMVNControlWrapper.dir/yarp_plugin_ixsensmvncontrol_wrapper.cpp.o
2019-12-04T11:55:58.5540234Z [51/53] Building CXX object wrappers/IWear/CMakeFiles/IWearWrapper.dir/src/IWearWrapper.cpp.o
2019-12-04T11:55:58.5540510Z ninja: build stopped: subcommand failed. 
~~~

**Expected behavior**
The compilation should be succssful. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04 
 - yarp version: 3.3.0
-  compiler: gcc 7.4.0

**Additional context**
This is the same content of https://github.com/robotology/yarp/issues/2118#issuecomment-561625727, I just opened a new issue to have a better visibility.

traversaro, 2019-12-05:
Interestingly, the project compiles fine before PR https://github.com/robotology/yarp/pull/2136 , thanks to the workaround added in https://github.com/robotology/wearables/pull/65/files .

drdanz, 2019-12-05:
Is it using `yarp_idl_to_dir` or `yarp_add_idl`?

I reverted some changes to `yarp_add_idl` that shouldn't have been there in the first place, to make it compatible with YARP 3.2, but I don't think anything changed in the `yarp_idl_to_dir` function

See also this comment
https://github.com/robotology/yarp/issues/2118#issuecomment-561190114

traversaro, 2019-12-05:
> Is it using `yarp_idl_to_dir` or `yarp_add_idl`?

Sorry, it is using `yarp_add_idl` (see https://github.com/robotology/wearables/blob/aa991c502895e7c954e302839f992bfdbcdb7325/msgs/thrift/CMakeLists.txt#L53), I used the wrong function in the issue title, thanks for noticing it.

drdanz, 2019-12-05:
I suggest to revert robotology/wearables#65
At some point I think that yarp_add_idl should use the namespace instead of the path, but when it will happen I'll try to add some deprecation method and/or some warning

traversaro, 2019-12-05:
> I suggest to revert [robotology/wearables#65](https://github.com/robotology/wearables/pull/65)
> At some point I think that yarp_add_idl should use the namespace instead of the path, but when it will happen I'll try to add some deprecation method and/or some warning

Ok, I guess we can try, but just to clarify: the current master branch of `wearables` (after the changes in https://github.com/robotology/wearables/pull/65) compiles fine with YARP 3.2 .

drdanz, 2019-12-06:
But did it compile with YARP 3.2 *without* those changes?

traversaro, 2019-12-06:
> But did it compile with YARP 3.2 _without_ those changes?

Yes.

lrapetti, 2019-12-16:
> I suggest to revert [robotology/wearables#65](https://github.com/robotology/wearables/pull/65)
> At some point I think that yarp_add_idl should use the namespace instead of the path, but when it will happen I'll try to add some deprecation method and/or some warning

reverting the PR seems to fix the compilation problem (https://github.com/robotology/wearables/pull/71).

traversaro, 2020-01-13:
I do not know if https://github.com/robotology/yarp/issues/1092 is related.

drdanz, 2020-07-14:
@traversaro @lrapetti sorry, I don't remember if we decided something regarding this issue... Did we decide to fix it or leave it like it is now?

traversaro, 2020-07-14:
> @traversaro @lrapetti sorry, I don't remember if we decided something regarding this issue... Did we decide to fix it or leave it like it is now?

I think that the regression is indeed there, but we already fixed our downstream projects, so I think we can simply close the issue.

closed 2020-07-14 by drdanz