robotology/yarp #2145: Streaming non yarp object through yarp ports

GiulioRomualdi, 2019-12-06:
I would like to stream a Vector (Matrix) through a `yarp::os::BufferedPort`, as far as I understood I could:
1. Store the data in a `yarp::sig::Vector` (`yarp::sig::Matrix`);
2. Define a custom *vector* that inherits from `yarp::os::Portable` class.

However `iDynTree`, `Eigen`, `openCV`, etc... define their vectors (Matrix). So, every time I would like to send the vector (matrix) through a `yarp::os::BufferedPort` I should instantiate a `yarp::sig::Vector` (`yarp::sig::Matrix`) and copy the value and this is time-consuming.

It would be nice to introduce a portable based on `Eigen::Map`, indeed every vector (matrix) can be easily wrapped using an `Eigen::Map<Eigen::MatrixXd>(.)` method. For instance, in `yarp` and `iDynTree` there are already functions that convert a Vector (Matrix) in `Eigen:Map<>` object.

What do you think? Is it worth spending time on this?
@traversaro @Nicogene @prashanthr05 

Related issue: https://github.com/robotology/yarp/issues/76

Nicogene, 2019-12-06:
Hi @GiulioRomualdi ,

You are right, in order to stream a custom type through the yarp network, that type should inherits from `yarp::os::Portable` classe, and you have to implement the methods `write()` and `read()`, where respectively you encode and decode respectively your custom data structure.

You are right, converting from a datastructure of another library could be time consuming, unless you simply copy the raw pointer of external data.

This has been possible in the `YARP_cv` library(see https://github.com/robotology/yarp/pull/1932) because exists the `setExternal` method in `yarp::sig::Image`.

On the other hand `yarp::sig::Vector` has not a method like this, and like `std::vector`, is a data structure that manage the memory, it doesn't use memory manage by external objects.

Actually https://github.com/robotology/yarp/pull/1582 was intended to add `setExternal` in `Vector`, but after a long discussion we decided that it would be better to implement a separate class called `VectorMap` that uses external memory, in order to make the memory management clear and explicity.
The idea was to do something similar to [`std::string_view`](https://en.cppreference.com/w/cpp/string/basic_string_view).

Unfortunately we never found the time to add this class that would be very useful indeed.

traversaro, 2020-09-10:
Related PR on iDynTree: https://github.com/robotology/idyntree/pull/734 . Having a similar `MatrixView` class also in YARP that inherits from `yarp::os::Portable` may be indeed quite convenient.

drdanz, 2020-09-11:
Related: #1603

I think all types in general require a big refactor. In order to split the "storage" part of the types from their serialization

closed 2020-12-22 by drdanz