robotology/yarp #2154: multipleanalogsensorsserver: open ports upon open()?

jgvictores, 2019-12-17:
**Is your feature request related to a problem? Please describe.**
In our use case, we usually instance devices with server-side network wrappers via `yarpdev`.  In the case of the wrappers we most use (namely `controlboardwrapper2`, `grabber`/`grabberDual`, `RGBDSensorWrapper`, and `analogServer`), when instanced via `yarpdev`, these wrappers open ports which enable client-side connections.
In the case of `multipleanalogsensorsserver` which I understand is a replacement of `analogServer`, no ports are open via the `yarpdev` mechanism.

**Describe the solution you'd like**
It would be great to enable `multipleanalogsensorsserver` to be compatible with our `yarpdev` workflow; this is, for `yarpdev --device multipleanalogsensorsserver --subdevice myanalogsensors` to open ports which enable client-side connections.

**Describe alternatives you've considered**
I understand that `yarpdev` calls the device `open()`, which in the case of `multipleanalogsensorsserver`, seems to not be sufficient. At time of writing, however, I have no knowledge of how to implement the logic I think I see in `controlboardwrapper2` (maybe?) for the code to reach the `attachAll()` section where ports seem to be actually open.

Thank you! :smiley:

Nicogene, 2019-12-18:
Hi @jgvictores,

The feature you are requesting is not difficult to implement, but here we should decide if the mechanism `device`-`subdevice` is something to deprecate/discourage or not.

It allows to run the couple server-client through `yarpdev` but it complicates the handling of the paramenter, making not clear which parameters are for the client, which are for the server.

Have you tried to add in the CMakeLists.txt of `myanalogsensors` the `EXTRA_CONFIG WRAPPER=multipleanalogsensorsserver` ?

It should you allow to run your device wrapped by `multipleanalogsensorsserver` throught the command `yarpdev --device myanalogsensors`.

jgvictores, 2019-12-18:
Hi @Nicogene,

Yes, the `EXTRA_CONFIG_WRAPPER` is present. Here is a small independent example so we can debug (note the plural to not collide with the robotology device wrapped by analogServer): https://github.com/jgvictores/fakeAnalogSensors

Running `yarpdev --device fakeAnalogSensors --name /sensors --period 40` provides the following output:
```bash
[INFO]created device <multipleanalogsensorsserver>. See C++ class MultipleAnalogSensorsServer for documentation.
yarp: Port /sensors/quit active at tcp://my.ip.goes.here:10002/
[INFO]device active in background...
```

Two phenomena can be observed:
1. Only the `/sensors/quit` port is open
2. The `open()` of the actual `fakeAnalogSensors` device is not reached

Imho it would be great for this device to work as other devices, so it would great if we could find a solution that meets all needs.

traversaro, 2019-12-18:
Hi @jgvictores, 
   thanks for the detailed issue. I think that back at the time of https://github.com/robotology/yarp/pull/1586/files I did not include the handling of the `subdevice` parameter, as I erroneously thought that it was handled at the `yarpdev`  or at the `yarp::dev::PolyDriver` level.

While I think it it makes little sense to duplicate the logic of handling the `subdevice` in all the network wrapper/server devices, and in general the `subdevice` parameter mechanism could be substituted with something more flexible, for uniformity and consistency with the existing YARP workflow, we should support the  `subdevice` parameter also in the `multipleanalogsensorsserver` device. This is necessary because the `subdevice` parameter is mentioned and used in the following points in the codebase: 
* https://github.com/robotology/yarp/blob/0ac66ed8b703eecd2207e24dbff4432008761b7b/src/libYARP_dev/src/yarp/dev/PolyDriver.cpp#L276
* https://github.com/robotology/yarp/blob/0ac66ed8b703eecd2207e24dbff4432008761b7b/doc/cmd_yarpdev.dox#L98

As @Nicogene mentioned, the modifications are relatively simple, something similar to https://github.com/robotology/yarp/pull/1018 and https://github.com/robotology/yarp/pull/1583 . I am short of time in the next period, so if anyone is willing to open a PR with these changes I would be happy to review it, thanks!

closed 2020-03-04 by drdanz