robotology/yarp #2159: Build failure on macOS

traversaro, 2019-12-19:
**Describe the bug**
The build of the master branch is broken on macOS 10.15 . 

**To Reproduce**
Build on macOS, there is a  compilation  failure: 
~~~
2019-12-19T02:37:52.8733920Z [ 70%] Building CXX object src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/Module.cpp.o
2019-12-19T02:37:53.5398600Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/YARP/src/yarprobotinterface/Module.cpp:74:25: error: member reference type 'struct __darwin_mcontext64 *' is a pointer; did you mean to use '->'?
2019-12-19T02:37:53.5399210Z     context->uc_mcontext.gregs[REG_RIP]++;
2019-12-19T02:37:53.5399320Z     ~~~~~~~~~~~~~~~~~~~~^
2019-12-19T02:37:53.5399710Z                         ->
2019-12-19T02:37:53.5420750Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/YARP/src/yarprobotinterface/Module.cpp:74:26: error: no member named 'gregs' in '__darwin_mcontext64'
2019-12-19T02:37:53.5421280Z     context->uc_mcontext.gregs[REG_RIP]++;
2019-12-19T02:37:53.5421390Z     ~~~~~~~~~~~~~~~~~~~~ ^
2019-12-19T02:37:53.5467020Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/YARP/src/yarprobotinterface/Module.cpp:74:32: error: use of undeclared identifier 'REG_RIP'
2019-12-19T02:37:53.5467550Z     context->uc_mcontext.gregs[REG_RIP]++;
2019-12-19T02:37:53.5467650Z                                ^
2019-12-19T02:37:53.5871150Z 3 errors generated.
~~~

**Expected behavior**
The compilation to be successful. 

**Configuration (please complete the following information):**
 - OS: macOS 10.15 
 - yarp version: https://github.com/robotology/yarp/commit/3580beeb14ae83d7534a9c3a764ca1d6a07ab9df
-  compiler: clang

**Additional context**
Link to the failed build in GitHub Actions: https://github.com/robotology/robotology-superbuild/runs/355635206 .

traversaro, 2019-12-19:
I guess the related commit is https://github.com/robotology/yarp/commit/93238fb6d3836fee59c467796e4c607fe18adea1 .

Nicogene, 2019-12-20:
Quite strange error, I will handle next week

traversaro, 2019-12-20:
> 
> 
> Quite strange error, I will handle next week.
> 
> `yarp-3.3` compiles?

Yes, the commit that introduced that code is https://github.com/robotology/yarp/commit/93238fb6d3836fee59c467796e4c607fe18adea1 , and it was not added to the `yarp-3.3` branch.

drdanz, 2019-12-20:
All the macos builds are now failing :boom: :rage: :boom:

drdanz, 2019-12-20:
Ups, I think that's my fault

drdanz, 2019-12-20:
Should be fixed by bc074eb6059140ddacc24b313da94b507d07ef05

closed 2019-12-23 by drdanz