robotology/yarp #2161: opencv_grabber should use YARP_cv utilities

Nicogene, 2019-12-20:
`opencv_grabber` converts the `cv::Mat` to a `yarp::sig::image` with a `memcpy`.

https://github.com/robotology/yarp/blob/0ac66ed8b703eecd2207e24dbff4432008761b7b/src/devices/opencv/OpenCVGrabber.cpp#L196-L200

We should use instead `fromCvMat` that not involves copies:

https://github.com/robotology/yarp/blob/0ac66ed8b703eecd2207e24dbff4432008761b7b/src/libYARP_cv/src/yarp/cv/Cv-inl.h#L90-L112

PeterBowman, 2020-08-27:
I'm afraid it is not viable, or at least not as straightforward as a simple invocation of `yarp::cv::fromCvMat`. Since no copies are involved, but rather `yarp::sig::Image` holds a pointer to a memory location managed by an instance of `cv::Mat`, the lifetime of the latter is critical. However, this `cv::Mat` dies as soon as the `getImage` method is exited, therefore the wrapping `grabberdual`  device attempts to forward a bogus `yarp::sig::Image` over the network (the memory location of the desired image may be erased at that point).

drdanz, 2021-10-05:
Fixed by @PeterBowman in #2713

PR #2713 (2021-10-05): Avoid copies between `cv::Mat` and `yarp::sig::Image` in opencv_grabber
Resolves #2161. In contrast to the suggested solution, I am converting a `yarp::sig::Image` to a `cv::Mat` so that final allocation occurs at the YARP caller side (doing it on a local `cv::Mat` would produce memory corruption as soon as the execution flow exits `getImage`) without using YARP_cv at all since color conversion must take place after the frame is grabbed. As a side note, I noticed in-place `cv::cvtColor` produces bogus results (frames flash between BGR and RGB representations), hence the auxiliary `bgr` variabl [... cut]
closed 2021-10-05 by drdanz