robotology/yarp #2164: Request tutorials on image processing

YANGGANG1971, 2019-12-25:
I want to obtain 3-D position information of green ball, how to do that? Do you have any tutorials?

pattacini, 2019-12-26:
Hi @YANGGANG1971 

Your request is not very clear. Which green ball are you referring to? Try to add as many details as possible in order to let us better understand what you're looking for.

My two cents is that you're currently tackling this assignment https://github.com/vvv-school/assignment_make-it-roll, where the ball is blue by the way 😉 

Am I right?
If that's correct, we can conveniently move this issue therein and try to give you hints on that.

YANGGANG1971, 2019-12-26:
You are right, I am tracking https://github.com/vvv-school/assignment_make-it-roll. Now it can run correctly. I just want to expand this research. In this tutorial, the blue ball position refers to the 2-D position. I want to know how to get the spatial information (i.e. 3-D position).

pattacini, 2019-12-26:
@YANGGANG1971 this issue has been moved to https://github.com/vvv-school/assignment_make-it-roll/issues/7.

closed 2019-12-26 by pattacini