robotology/yarp #2167: Installation instructions for yarpmod realsense2 on Win32

randaz81, 2020-01-02:
To install the module `realsense2`, user needs to set the environment variable `realsense2_DIR`.
However, no instructions are provided about what this variable should point to, on both linux and windows.

- On windows, if you install the _realsense-sdk_ from https://github.com/IntelRealSense/librealsense/releases, yarp does not detects the library, even if realsense2_DIR is properly set to the installation folder (e.g. _C:\Program Files (x86)\Intel RealSense SDK 2.0_). I haven't tested this installation procedure on Linux, even if it is available.
- if you clone the repository from https://github.com/IntelRealSense/librealsense, you build it, and you set realsense2_DIR to the build dir, **yarp detects the library but the ccmake fails afterwards**, complaining about missing headers (the same error is shown on Linux too). This is pretty misleading.
- the solution that worked for me was to **install the library**, after building it, making realsense2_DIR pointing to the installed directory.

Nicogene, 2020-01-02:
Hi @randaz81!

It seems to me more a problem that should be resolved upstream by Intel if the SDK installer doesn't install the library and include headers where their `realsense2Config.cmake` try to find it.

Following their installation instruction on linux/macOs, `find_package(realsense2)` works fine.

I suggest to report them these problems on windows if no one has done it yet.

abhi604, 2020-01-23:
that was very good