robotology/yarp #2179: Introduce standardized methods for retrieving const_iterator in VectorOf

GiulioRomualdi, 2020-01-14:
In order to access a const iterator of a `yarp::sig::vector`  the methods [`cbegin()`](http://www.yarp.it/classyarp_1_1sig_1_1VectorOf.html#a8f2440269480a9675d0b5e7e6d5e2415)/[`cend()`](http://www.yarp.it/classyarp_1_1sig_1_1VectorOf.html#a570ada22c397bed321d28729f5a58502) have to be used. Renaming the methods by removing the `c` may be useful when template functions are implemented. I.e. std:: vector<> uses the `begin()`/`end()` to access to const iterators.

The new methods can be easily implemented as follows
```patch
diff --git a/src/libYARP_sig/src/yarp/sig/Vector.h b/src/libYARP_sig/src/yarp/sig/Vector.h
index 016f354d3..9e13e4c16 100644
--- a/src/libYARP_sig/src/yarp/sig/Vector.h
+++ b/src/libYARP_sig/src/yarp/sig/Vector.h
@@ -550,6 +550,22 @@ public:
         return first + len;
     }
 
+    /**
+     * @brief Returns a const iterator to the beginning of the VectorOf
+     * @note At the moment iterator is implemented as a pointer, it may change in the future.
+     * For this reason it should not be used as a pointer to the data, use data() instead.
+     */
+    const_iterator begin() const noexcept {
+        return first;
+    }
+
+    /**
+     * @brief Returns a const iterator to the end of the VectorOf.
+     */
+    const_iterator end() const noexcept {
+        return first + len;
+    }
+
     /**
      * @brief Returns a const iterator to the beginning of the VectorOf
      * @note At the moment iterator is implemented as a pointer, it may change in the future.
```

What do you think? @drdanz @traversaro

traversaro, 2020-01-15:
I think it makes sense, probably it could make sense to also deprecate cbegin and cend if this methods are implemented.

Nicogene, 2020-01-15:
Actually in std::vector begin/end cbegin/cend coexist see https://en.cppreference.com/w/cpp/container/vector/begin.

I will keep all the methods without deprecating it.

traversaro, 2020-01-15:
> 
> 
> Actually in `std::vector` `begin/end-cbegin/cend` coexist see https://en.cppreference.com/w/cpp/container/vector/begin.
> 
> I will keep all the methods without deprecating it.

I was not aware of this, yes then it make sense to keep both.

GiulioRomualdi, 2020-01-15:
Me neither.  If you (@nicogene) want, I can handle the PR in which `begin()` / `end()` are introduced

Nicogene, 2020-01-15:
@GiulioRomualdi go ahead :+1:

drdanz, 2020-01-15:
:+1: For adding const `begin` and `end`, :-1: for deprecating `cbegin` and `cend` which are required to return a const iterator when you have a non-const object

drdanz, 2020-03-11:
Fixed by @2182

closed 2020-03-11 by drdanz