robotology/yarp #2180: How to properly install OpenCV dependency for yarp on Windows?

randaz81, 2020-01-14:
My system is Windows 10, using Microsoft Visual Studio Enterprise 2019
Version 16.4.3.

I downloaded OpenCV 3.4.2 from https://sourceforge.net/projects/opencvlibrary/files/opencv-win/3.4.2/opencv-3.4.2-vc14_vc15.exe/download  (the same version I was using on my old laptop). 

I configured the sources with cmake, I built it successfully (Debug and Release) and installed under the path: `C:/Software/yarp_software/robotology_install/opencv3.4/sources/build/install`, then I set the environment variable `OpenCV_DIR` to the aforementioned directory.

When trying to configure yarp with cmake, I get the following warning and all the projects related to OpenCV are turned off.
```
CMake Warning at C:/Software/yarp_software/robotology_install/opencv3.4/sources/build/install/OpenCVConfig.cmake:156 (message):
  Found OpenCV Windows Pack but it has no binaries compatible with your
  configuration.

  You should manually point CMake variable OpenCV_DIR to your build of OpenCV
  library.
Call Stack (most recent call first):
  extern/ycm/ycm-0.11.0/find-modules/FindOpenCV.cmake:117 (include)
  cmake/YarpFindDependencies.cmake:321 (find_package)
  CMakeLists.txt:61 (include)
```

I'm not sure about what should I do...

randaz81, 2020-01-15:
Update: if follow the same identical procedure using Opencv4.2  https://sourceforge.net/projects/opencvlibrary/files/4.2.0/opencv-4.2.0-vc14_vc15.exe/download 
everything is ok.

Maybe Opencv 3.4.2 is just too old? Which are the supported/recommended versions?

randaz81, 2020-01-15:
Update2: also Opencv 3.9 works properly. So maybe the problem is related only to the old Opencv 3.4.2.

**PS: I'm not sure about which version is included in robotology installer, however the link shown at http://wiki.icub.org/wiki/Windows:_installation_from_sources#Getting_iCub.27s_dependencies  is certainly outdated.**

traversaro, 2020-01-15:
The version that we recommend to use in robotology-superbuild is 3.4.4, see https://github.com/robotology/robotology-superbuild#yarp-icub--and-robotology-additional-dependencies-installers .

traversaro, 2020-01-15:
> PS: I'm not sure about which version is included in robotology installer, however the link shown at http://wiki.icub.org/wiki/Windows:_installation_from_sources#Getting_iCub.27s_dependencies is certainly outdated.

See https://github.com/robotology/icub-packaging/issues/10 for a related issue.

traversaro, 2020-02-19:
The new recommended way for installing dependencies for the robotology-superbuild on Windows is documented in https://github.com/robotology/robotology-superbuild/pull/348 .

closed 2020-02-25 by drdanz