robotology/yarp #2183: Rethink-refactor the way the camera/depth devices are handled in YARP

Nicogene, 2020-01-15:
Talking with @randaz81 and @xEnVrE emerged the necessity more "composability" (@traversaro :stuck_out_tongue_closed_eyes: ) for how the camera/depth sensors are handled.

The [`yarp::dev::IRGBDSensors`](http://yarp.it/classyarp_1_1dev_1_1IRGBDSensor.html) interface and [RGBDSensorWrapper](http://yarp.it/classyarp_1_1dev_1_1RGBDSensorWrapper.html) served well but it seems to be not enough for the necessities that came out.

The necessity is to handle a great variety of sensors (e.g. the various realsense cameras) that have different streams available.
For example some have stereo cameras, some depth sensors, some others are RGBD sensors, some have IMU etc.

My idea was to adopt a structure similar to the [MultipleAnalogInterfaces](http://yarp.it/group__dev__iface__multiple__analog.html).

This issue is for discuss about this topic, and start a first implementation.

cc @vtikha @aerydna @drdanz 
@barbalberto would be great having your opinion :heart:

traversaro, 2020-01-15:
cc @prashanthr05

barbalberto, 2020-01-15:
I'd be happy to give my opinion, but I need more information. 
Can you provide some example of sensors which arised this issue?
What exactly is not achievable right now?

aerydna, 2020-01-17:
i helped @barbalberto develop the interface and for what i remember it seems versatile enough to handle the things mentioned. Maybe we should discuss this F2F

Nicogene, 2020-02-06:
> I'd be happy to give my opinion, but I need more information.
Can you provide some example of sensors which arised this issue?
What exactly is not achievable right now?

Sorry @barbalberto this discussion fell in low priority for now, but it will be done for sure in the near future (I hope). I will write to you a proper report/proposal about the problem :heart:

closed 2020-12-22 by drdanz