robotology/yarp #2189: Conditional jump or move depends on uninitialised value(s) regression in YARP 3.3.101

traversaro, 2020-01-23:
**Describe the bug**
We have a regression in a iDynTree test, see: 
~~~
==25176== Conditional jump or move depends on uninitialised value(s)
==25176==    at 0x5A57EAC: yarp::os::ManagedBytes::clear() (in /home/runner/work/idyntree/idyntree/install/lib/libYARP_os.so.3.3.101)
==25176==    by 0x5A57F30: yarp::os::ManagedBytes::moveOwnership(yarp::os::ManagedBytes&) (in /home/runner/work/idyntree/idyntree/install/lib/libYARP_os.so.3.3.101)
==25176==    by 0x4E561E8: iCub::iDynTree::DynTree::setAng(yarp::sig::VectorOf<double> const&) (in /home/runner/work/idyntree/idyntree/build/lib/libidyntree-yarp-kdl.so.1.0.100)
==25176==    by 0x10E1F7: set_random_q_dq_ddq(yarp::os::Random&, iCub::iDynTree::DynTree&) (in /home/runner/work/idyntree/idyntree/build/bin/iDynTreeCOMTest)
==25176==    by 0x10B5CD: main (in /home/runner/work/idyntree/idyntree/build/bin/iDynTreeCOMTest)
~~~

**To Reproduce**
I still need to investigate how to reproduce the problem, but I preferred to report it nevertheless. 

**Expected behavior**
Valgrind should not complain.

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04 
 - yarp version: https://github.com/robotology/yarp/commit/ddb57bcc7b0f19d4187f4bede4d05b601dfe1a42
-  compiler: `gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0`

**Additional context**
See https://github.com/robotology/idyntree/commit/98c274ea62cde872dd70bd0c713f820a485f4d74/checks?check_suite_id=416035422#step:13:212 .

traversaro, 2020-01-23:
This may be a regression of https://github.com/robotology/yarp/pull/1974 .

traversaro, 2020-01-24:
Output with Debug enabled: 
~~~
106: Test command: /usr/bin/valgrind "--leak-check=full" "--error-exitcode=1" "/home/straversaro/src/robotology-superbuild/build/robotology/iDynTree/bin/iDynTreeCOMTest"
106: Test timeout computed to be: 1500
106: ==25720== Memcheck, a memory error detector
106: ==25720== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
106: ==25720== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
106: ==25720== Command: /home/straversaro/src/robotology-superbuild/build/robotology/iDynTree/bin/iDynTreeCOMTest
106: ==25720== 
106: ==25720== Conditional jump or move depends on uninitialised value(s)
106: ==25720==    at 0x618B16D: yarp::os::ManagedBytes::clear() (ManagedBytes.cpp:166)
106: ==25720==    by 0x618AC1D: yarp::os::ManagedBytes::moveOwnership(yarp::os::ManagedBytes&) (ManagedBytes.cpp:62)
106: ==25720==    by 0x618ACDF: yarp::os::ManagedBytes::ManagedBytes(yarp::os::ManagedBytes&&) (ManagedBytes.cpp:73)
106: ==25720==    by 0x5EA967C: yarp::sig::VectorOf<double>::VectorOf(yarp::sig::VectorOf<double>&&) (Vector.h:192)
106: ==25720==    by 0x5EA8930: yarp::sig::Matrix::subcol(unsigned long, unsigned long, unsigned long) const (Matrix.cpp:428)
106: ==25720==    by 0x50DF5B1: YarptoKDL(yarp::sig::Matrix const&, KDL::Frame&) (yarp_kdl.cpp:32)
106: ==25720==    by 0x504F84D: iCub::iDynTree::DynTree::setWorldBasePose(yarp::sig::Matrix const&) (DynTree.cpp:277)
106: ==25720==    by 0x10D9C1: set_random_q_dq_ddq(yarp::os::Random&, iCub::iDynTree::DynTree&) (iDynTreeCOMTest.cpp:141)
106: ==25720==    by 0x10E0F5: main (iDynTreeCOMTest.cpp:205)
106: ==25720== 
106: iDynTree Jacobian test world pose  1.000000	 0.000000	 0.000000	 0.000000
106:  0.000000	 1.000000	 0.000000	 0.000000
106:  0.000000	 0.000000	 1.000000	 0.000000
106:  0.000000	 0.000000	 0.000000	 1.000000
106: ==25720== Conditional jump or move depends on uninitialised value(s)
106: ==25720==    at 0x618B16D: yarp::os::ManagedBytes::clear() (ManagedBytes.cpp:166)
106: ==25720==    by 0x618AC1D: yarp::os::ManagedBytes::moveOwnership(yarp::os::ManagedBytes&) (ManagedBytes.cpp:62)
106: ==25720==    by 0x618ACDF: yarp::os::ManagedBytes::ManagedBytes(yarp::os::ManagedBytes&&) (ManagedBytes.cpp:73)
106: ==25720==    by 0x505DCAB: yarp::sig::VectorOf<double>::VectorOf(yarp::sig::VectorOf<double>&&) (Vector.h:192)
106: ==25720==    by 0x504FC7B: iCub::iDynTree::DynTree::setAng(yarp::sig::VectorOf<double> const&) (DynTree.cpp:358)
106: ==25720==    by 0x10DB28: set_random_q_dq_ddq(yarp::os::Random&, iCub::iDynTree::DynTree&) (iDynTreeCOMTest.cpp:148)
106: ==25720==    by 0x10E0F5: main (iDynTreeCOMTest.cpp:205)
106: ==25720== 
~~~

drdanz, 2020-01-24:
My fault, I added `clear()` in `moveOwnership` to fix the memory leak, but it should have probable been added to the move operator only.
In this way this is executed in the move constructor.

Also this demonstrates that there are no relative unit tests... 

I will fix it ASAP.

traversaro, 2020-01-24:
> Also this demonstrates that there are no relative unit tests...

I am adding a regression tests for this now.

traversaro, 2020-01-24:
Minimal regression tests added in https://github.com/traversaro/yarp/commit/6cc5dd5f869eec82e3c639e84e5c33d21d379a01 . As the error disappears if you remove the`if (false) { return Vector(0); } `, I guess you can reproduce it only in Debug mode.

traversaro, 2020-01-24:
Output of the regression tests: 
~~~
==8929== Conditional jump or move depends on uninitialised value(s)
==8929==    at 0x579D16D: yarp::os::ManagedBytes::clear() (ManagedBytes.cpp:166)
==8929==    by 0x579CC1D: yarp::os::ManagedBytes::moveOwnership(yarp::os::ManagedBytes&) (ManagedBytes.cpp:62)
==8929==    by 0x579CCDF: yarp::os::ManagedBytes::ManagedBytes(yarp::os::ManagedBytes&&) (ManagedBytes.cpp:73)
==8929==    by 0x184AAE: yarp::sig::VectorOf<double>::VectorOf(yarp::sig::VectorOf<double>&&) (Vector.h:192)
==8929==    by 0x17E40A: functionThatReturnsCopyOfReference(yarp::sig::VectorOf<double> const&) (VectorTest.cpp:149)
==8929==    by 0x18287A: ____C_A_T_C_H____T_E_S_T____0() (VectorTest.cpp:389)
==8929==    by 0x19B471: Catch::TestInvokerAsFunction::invoke() const (catch.hpp:12209)
==8929==    by 0x19AAE0: Catch::TestCase::invoke() const (catch.hpp:12107)
==8929==    by 0x1951F3: Catch::RunContext::invokeActiveTestCase() (catch.hpp:10937)
==8929==    by 0x194F38: Catch::RunContext::runCurrentTest(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) (catch.hpp:10910)
==8929==    by 0x193975: Catch::RunContext::runTest(Catch::TestCase const&) (catch.hpp:10680)
==8929==    by 0x196704: Catch::(anonymous namespace)::runTests(std::shared_ptr<Catch::Config> const&) (catch.hpp:11242)
==8929==  Uninitialised value was created by a stack allocation
==8929==    at 0x17E457: ____C_A_T_C_H____T_E_S_T____0() (VectorTest.cpp:153)
==8929== 
~~~

closed 2020-01-25 by drdanz

mobbinglikeathug, 2020-02-01:
> set_random_q_dq_ddq(yarp::os::Random

commit 159ebb1a (2020-01-25): Fix "Conditional jump or move depends on uninitialised value(s)"

Fixes #2189
commit ef19d8e5 (2020-01-25): Merge pull request #2190 from drdanz/fix_2189

Fix #2189