robotology/yarp #2196: Question: Does yarpscope allow auto y-axis scaling

Yeshasvitvs, 2020-02-07:
**Is your feature request related to a problem? Please describe.**

Is there a feature in yarpscope that allows for auto y-axis scaling? Currently, I am using `minvalue` and `maxvalue` parameters.

**Describe the solution you'd like**

It would be nice to have an auto scaling option for y-axis values which will get scaled based on the values streamed through the yarp port.

@drdanz

pattacini, 2020-02-07:
cc @Nicogene 😉

Yeshasvitvs, 2020-02-07:
Thanks @pattacini

Nicogene, 2020-02-10:
Thanks for the trust @pattacini but I am not a great expert of `yarpscope`, I will bounce it back to @drdanz :smile:

pattacini, 2020-02-10:
@Nicogene I had only replaced another user with you - as there was a wrong mentioning.

drdanz, 2020-02-10:
Sorry, no idea, and I didn't write the code for the qt version...

Yeshasvitvs, 2020-02-12:
@drdanz who would be the right person to check this issue ?

drdanz, 2020-02-12:
Unfortunately, we commissioned it to a third party long time ago. I don't think there is a "maintainer" right now...